bambulab/BambuStudio #1141: Menu  File - New Window - does not open

Manfredo1952, 2023-01-21:
**Describe the bug**
I have a window open and an object on the plate.
I click in the top menu  File / New Window - nothing happens
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to File, New Window
2. Click on New Window
3. nothing happens

**Expected behavior**
A new window should lopen ?

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS:  MacOS
 - Version 11.7.3
 -

bambu123, 2023-01-28:
I had a check and can't reproduce here, and there is no relate logs in latest version..
Can you check help your environment:
1. What version of Bambu Studio are you using, is that ok with the previous version of Bambu Studio ?
2. Have you installed Bambu Studio to Application and start with launch icon ?
3. Can this command line open an new instance of Bambu Studio: 'open -n -a BambuStudio'

Manfredo1952, 2023-01-28:
@bambu123   thanks for replying
1.  installed Version # is 1.4.2.13  -   remember that previous version could open new window as well (can't reproduce here right now)
2.  yes
3.  it will start a new instance, not a new window - but there is some crash window opening with this report, but I can close the crash report and keeo bioth instances alive.
4. by the way, When I click "new project" there is also no reaction

[image]

crash report from (3) follows here

Process:               BambuStudio [7370]
Path:                  /Applications/BambuStudio BL.app/Contents/MacOS/BambuStudio
Identifier:            BambuStudio
Version:               01.04.02.13 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           BambuStudio [7370]
User ID:               501

Date/Time:             2023-01-28 05:28:10.553 +0100
OS Version:            macOS 11.7.3 (20G1113)
Report Version:        12
Anonymous UUID:        5643E8CB-94AE-8792-AD62-5C62F09A8907

Sleep/Wake UUID:       15ADAAB3-495D-4860-B135-9A0A80A65A86

Time Awake Since Boot: 690000 seconds
Time Since Wake:       17000 seconds

System Integrity Protection: enabled

Crashed Thread:        22

Exception Type:        EXC_GUARD
Exception Codes:       0x6000000000000012, 0x0000000000000002
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    LIBSYSTEM, [0x2]

Thread 0:: bambustu_main
0   libc++.1.dylib                	0x00007fff207f0081 std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::__init_copy_ctor_external(char const*, unsigned long) + 29
1   com.bambulab.bambu-studio     	0x000000010ef0436a std::__1::pair<std::__1::__tree_iterator<std::__1::__value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > > >, std::__1::__tree_node<std::__1::__value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > > >, void*>*, long>, bool> std::__1::__tree<std::__1::__value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > > >, std::__1::__map_value_compare<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > > >, std::__1::less<void>, true>, std::__1::allocator<std::__1::__value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > > > > >::__emplace_unique_key_args<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::piecewise_construct_t const&, std::__1::tuple<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&>, std::__1::tuple<> >(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::piecewise_construct_t const&, std::__1::tuple<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&>&&, std::__1::tuple<>&&) + 74
2   com.bambulab.bambu-studio     	0x000000010ef00f74 bool nlohmann::detail::parser<nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > >, nlohmann::detail::input_stream_adapter>::sax_parse_internal<nlohmann::detail::json_sax_dom_parser<nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char> > > > >(nlohmann::detail::json_sax_dom_parser<nlohmann::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_trait [... cut ...]
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: APPLE HDD ST1000DM003, 1 TB
Serial ATA Device: APPLE SSD SM0128G, 121,33 GB
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: USB Audio Device
USB Device: Keyboard Hub
USB Device: Apple Keyboard
Thunderbolt Bus: iMac, Apple Inc., 26.1

bambu123, 2023-01-28:
@Manfredo1952 Currently we only support New Instances instead of New Windows. So the 'New Window' menu starts a new instance.
And the menu executes the following command:
/usr/bin/open -n -a /Applications/BambuStudio.app --args

Please help check whether this command works in your console.

Manfredo1952, 2023-01-28:
@bambu123   
Thanks, yes. It does open a new instance.  I am getting that crasj´h window again, but that can be deleted/closed and I can continue with the new instance.

bambu123, 2023-01-28:
The app path in your system is so strange.
/Applications/BambuStudio BL.app/Contents/MacOS/BambuStudio
instead of
/Applications/BambuStudio.app/Contents/MacOS/BambuStudio

This could be the reason. Have you selected the 'Keep Both' operation when installing the new version?

Manfredo1952, 2023-01-28:
@bambu123  I have just renamed the app resp. added BL to distiinguis with the SoftFever version-

bambu123, 2023-01-29:
Sorry, the 'New Window' menu can't work in this condition.

closed 2023-03-02 by SaltWei