bambulab/BambuStudio #1142: P1p flow calibration does not work with German language. If you switch to English it works.

n4mb3r0n3, 2023-01-21:
**Describe the bug**
A clear and concise description of what the bug is.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

theogott, 2023-01-23:
You mean we should just switch printer to english and all works fine?
Maybe i should also do that.

lanewei120, 2023-01-24:
thanks for reporting this issue

we have fixed it in local , and it will be ready in next version

currently please switch to English for calibration if needed,
and switch back, then input the k value got from calibration in English

Sorry for the inconvenience

holidaay, 2023-02-19:
Same problem with swedish. Works with english but not in swedish.

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei