bambulab/BambuStudio #1147: P1P (01.01.01.00) ignores seam placement and sets to random after slicing and sending to printer

sisuwest, 2023-01-22:
**Describe the bug**
Regardless of the seam placement I set in the slicer (aligned) the printer ignores that changes it to random. It doesn't show that in the slicer preview, but upon printing multiple zits are shown across a random seam. 

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
[image]
[image]

 - OS: macOS 
 - Version 13.1

RobinWhitfield, 2023-01-22:
It is impossible for the printer to change the location of the seam, so if there is indeed an issue with the seam selection being ignored, it's a slicer issue.

QingZhangBambu, 2023-01-28:
could you provide the 3mf file? I'm trying to reproduce that.

QingZhangBambu, 2023-08-01:
Some guys reproduced that, and we found that the calibration did not work well, PA and flow ratio were not right. 
This issue be close for no responsed

closed 2023-08-01 by QingZhangBambu