bambulab/BambuStudio #1148: libbambu_networking.so: crash in SsdpDiscovery on suspend/resume

jwise, 2023-01-22:
When I suspend/resume my laptop, Bambu Studio immediately crashes on the way back up, with the message: `*** buffer overflow detected ***: terminated`.

Running Bambu Studio in `gdb`, a backtrace is as follows:

```
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140735604774464) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140735604774464) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140735604774464, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007fffef7f0476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007fffef7d67f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007fffef8376f6 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7fffef989943 "*** %s ***: terminated\n") at ../sysdeps/posix/libc_fatal.c:155
#6  0x00007fffef8e476a in __GI___fortify_fail (msg=msg@entry=0x7fffef9898e9 "buffer overflow detected") at ./debug/fortify_fail.c:26
#7  0x00007fffef8e30c6 in __GI___chk_fail () at ./debug/chk_fail.c:28
#8  0x00007fffef8e46ab in __fdelt_chk (d=<optimized out>) at ./debug/fdelt_chk.c:25
#9  0x00007fffdc683da7 in BBL::SsdpDiscovery::ssdp_thread() () at /home/joshua/.config/BambuStudio/plugins/libbambu_networking.so
#10 0x00007fffdc939b3b in thread_proxy () at /home/joshua/.config/BambuStudio/plugins/libbambu_networking.so
#11 0x00007fffef842b43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#12 0x00007fffef8d4a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

This is inside the proprietary bambu_networking blob, so I can't tell *too* much about what's going on.  But I did a little triage.  The `buffer overflow detected` message in `fdelt_chk` comes from an attempt to do `FD_SET` or `FD_CLR` with a `fd` that is out of range -- from the man page:

> An fd_set is a fixed size buffer. Executing FD_CLR() or FD_SET() with a value of fd that is negative or is equal to or larger than FD_SETSIZE will result in undefined behavior. Moreover, POSIX requires fd to be a valid file descriptor. 

When I went to look at exactly what was going on, it became somewhat more obvious -- the failing check was here:

```
   0x00007fffdc683d98 <+88>:	rep stos %rax,%es:(%rdi)
   0x00007fffdc683d9b <+91>:	movslq 0x3e0(%rbx),%rdi
=> 0x00007fffdc683da2 <+98>:	call   0x7fffdc627040 <__fdelt_chk@plt>
```

Looking at the `fd` value that would have gotten passed to `__fdelt_chk` (being sign-extended from 32 bits to 64 bits):

```
(gdb) x/4bx $rbx + 0x3e0
0x61d000137860:	0xff	0xff	0xff	0xff
```

So it looks like something is trying to do a `FD_SET` or `FD_CLR` operation on a `fd` that is actually `-1`.

No idea where that comes from but it might make sense to you :) Maybe if I have some time I will disassemble more into `libbambu_networking` to see exactly which syscall returned `-1` before we blew up, just for some personal entertainment.

In the mean time, while chasing this down, I compiled BBS with address sanitizer and found a few bugs that way.  I'll submit a PR for those soon.

jwise, 2023-01-28:
Thanks.  Let me know if you need help reproducing the issue.  I can reliably reproduce it on my machine.

jwise, 2023-02-03:
Thanks, I will try it tonight!

jwise, 2023-02-04:
@StoneLiBambulab Looks good in this version.  Thanks!

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei