bambulab/BambuStudio #1151: X-Y hole/contour compensation ignored for painted objects

rehmi, 2023-01-22:
Steps to reproduce the behavior:
1. Unzip and open [hole compensation test.3mf.zip](https://github.com/bambulab/BambuStudio/files/10475228/hole.compensation.test.3mf.zip). Switch to the Prepare tab and zoom into the center of the build plate, where you should find three instances of the test object ([hole compensation test 5x5mm, 2mm diam.stl.zip](https://github.com/bambulab/BambuStudio/files/10475306/hole.compensation.test.5x5mm.2mm.diam.stl.zip)) as seen below:
[image]

2. Switch to the Preview tab or press the Slice button to view the sliced results:
[image]

The expected behavior would be for the toolpath of the painted, compensated object to resemble the unpainted, compensated object; instead, it resembles the unpainted, uncompensated object.

Tested with BambuStudio version 1.4.2.13 on macOS Ventura 13.1 (arm64).

SaltWei, 2023-01-28:
Thank you very much. We will have a check.

zhimin-zeng-bambulab, 2023-02-21:
The XY size compensation cannot be combined with multi-material painting, we will add a warning to prompt the user.
It is the same for Prusa.

rehmi, 2023-02-22:
Thank you for looking into this and planning to add a warning. Before closing this issue, could you say a bit more about the blocker? Presumably this is a limitation imposed by the current slicer architecture that you learned by exploring the matter in some depth; it would benefit future generations if you could share your discoveries before moving on to other matters.

zhimin-zeng-bambulab, 2023-02-22:
The current software framework does not support it.
In short, the coloring information is recorded on the object before size compensation, and it cannot be found after size compensation.

closed 2023-03-02 by SaltWei