bambulab/BambuStudio #1152: Export generated support as STL after slicing

hutje, 2023-01-22:
When you are bound to another slicer because the printer only allows their own slicer, it can't use the nice feature of the way BambuStudio handles the supports like the slim tree or hybrid support

It would be nice to have an option to export the supports as separate STL to be able to import it as an object in the other slicer.

theogott, 2023-01-23:
If you can import and export the Supports as STL, it would even be possible to make changes on these, for example in Blender.
IMPORT & EXPORT Supports AS STL-File (without the Object)?
Sounds good.

reefball, 2024-03-26:
I would VERY VERY much like it if we could export the supports as an STL file so they could be customized.  This would be an AMAZING feature...if anyone figures out how to do this PLEASE let me know.

BearThreat, 2024-10-30:
one click auto generated supports and then export as an stl is a huge pain saving feature. Everyone who doesn't want to hand build tree supports would download BambuStudio. There are years of related questions on google and reddit. I just went down a long rabbit hole searching for ANY slicer that exports to stl. Some resin slicers allow tree support + and stl exporting, but the resin style tree supports don't translate well to fdm printers.