bambulab/BambuStudio #1154: Move flow calibration lines between prints - printing them in the same spot destroys build plates

c2h5oh, 2023-01-23:
**Is your feature request related to a problem? Please describe.**
All calibration lines are always printed in the exact same spot of the build plate. This results in build plates wearing much faster in those spots: all 3 build plates I own have a hole in sticker where calibration line starts (or ends?), because there is a slightly larger blob on that end that sticks to plates hard regardless of gluestick.

**Describe the solution you'd like**
Move calibration lines within designated area between prints (e.g. define several patterns and select one based on print hash or select one at random when slicing) to more evenly wear the build plates and extend their life.

theogott, 2023-01-23:
Hmm, I have printed a lot (see picture) and i can not confirm this, while lately i only use the PEI Plate where its not a issue at all.
While i would like such a Feature, there are only 3 alternative positions for the Patterns.

[image]

c2h5oh, 2023-01-23:
Textured plate is perfect. Two hot and one cool plate after 5~10 spools worth of small prints each have sticker damage in my case.

npoulos, 2023-01-24:
or even allow to move the lines close to the front of the plate. So much space is lost to the calibration lines.

tangketan, 2023-03-02:
The calibration lines are defined in start gcode and difficult to change. We'll figure out a way to solve the original issue "same spot wears out build plates".

SaltWei, 2024-06-05:
PEI plate has no issue, and it is the default plate for printer now.

closed 2024-06-05 by SaltWei