bambulab/BambuStudio #1157: Wasted time with multi-colours print and flush into objects

OmiCron07, 2023-01-23:
I'm doing a two-colour print with 7 objects, of which 3 have checked the "flush into this object" option. So, I have minimal or no purging, but it seems it wastes 10-20 seconds for nothing.

1. The filament is cut.
2. The filament is retracted.
3. The new filament is loaded.
4. The temperature is changed from 220 to 240 and waiting for the set temperature.
5. The purging process starts.
6. The temperature is changed back to 220 and waiting to cool.

But step 5 seems to do nothing because it will purge in my objects and not in the chute. So, we are wasting time just to heat up and immediately cool down. What I have in my purge bin is only filament hairs, like only the drooling filament because we heat it up at 240.

[image]
[image]

Some math. If the useless process takes 15 seconds and there are 1000 colour changes, it takes a whole 4 hours just waiting for nothing.

What I'm printing: [Hilt.zip](https://github.com/bambulab/BambuStudio/files/10483694/Hilt.zip)

Videos trying to show the issue: 
https://photos.app.goo.gl/PP7zFuiCaGhzd6LSA
https://photos.app.goo.gl/PP7zFuiCaGhzd6LSA

zhimin-zeng-bambulab, 2023-01-29:
Thanks for reporting this issue！
We will optimize the purging process.

zhimin-zeng-bambulab, 2023-01-31:
The optimized logic is that when the flushing volume is 0, there will be no heating up and cooling down. When the flushing volume is greater than 0, the original logic is still maintained.

SaltWei, 2023-03-02:
can be checked on v1.5.0 Public beta