bambulab/BambuStudio #1159: Automatic Filament switching doesnt work

randyre, 2023-01-23:
**Describe the bug**
Automatic Filament switching on Runout detection won't allow continue.  #1096 

Wants me to put filament into empty slot, but I'm remote and am trying to get it to switch to any other PLA slot of any color and it won't allow me to load, only to continue, which then wants me to load the now empty slot.

Currently load and unload and changing slots is prohibited while printer is printing. Even while paused

**Expected behavior**
One of the reasons for having an AMS is to allow continued printing. Would like to allow changing slots and loading and unloading while print is paused.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version Current

StoneLiBambulab, 2023-01-24:
Hi @randyre ,
   "load" and "unload"  are only used for non-printing now. When filament is run out, we need to complement the user interaction. For example, let the user choose a different slot to print or replace the same filament. 
There is currently only a default behavior in the printer. We'll add the iteraction as soon as we discuss it.

Thanks!

Stone Li

michalss, 2023-01-29:
same here. It wont do anything, stuck and you cannot even load a new filament...

michalss, 2023-01-29:
Another issue is even sensor see there is not fillamen in ASM and trying to change it, it wont let you a load a new one... :( very sad

camojackson, 2023-02-28:
Had this error this morning. I had two black spools of PLA loaded in expecting one to runout overnight. Woke up to a print error and had to move the fresh spool from its slot to the now empty slot. Both had all the same identifiers and colors selected in the AMS and reflected in the slicer.

walterwongbbl, 2023-05-25:
Hi.randyre
1. Firstly, the material type and color need to be the same in order to be allowed for Automatic Filament switching
2. There were some issues with the previous printer firmware and it needs to be updated to version 01.05.01.00
3. In the next version, you can see the details of automatic filament switching 
I hope this can help you

StoneLiBambulab, 2023-10-24:
Close this issue as fixed, open a new one if there is any problem.

Thanks!

Stone Li

closed 2023-10-24 by StoneLiBambulab