bambulab/BambuStudio #1161: Flush into object should be restricted when using Support-W

randyre, 2023-01-23:
**Describe the bug**
Flush into material is selected and when using support materials this causes a structurally weak or failed part.

**To Reproduce**
Steps to reproduce the behavior:
1. Print object with support-w enabled as support material or support interface.
2. Complete print and print will have bad structural integrity or problems with layer adhesion.

**Expected behavior**
Multi material prints should either not allow flush in infill when support-w is used or not allow support-w to flush while allowing other materials to flush.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version ALL

Benderrodrigu3z, 2023-01-24:
How do you still have Support W ? that's been out of stock in the store forever

randyre, 2023-02-04:
> How do you still have Support W ? that's been out of stock in the store forever
@Benderrodrigu3z 
One came with X1C from Kickstarter and bought another when they came back in stock, and then use it **VERY** sparingly, mainly for things that either need a near perfect surface when supports are removed, or where it's really hard to remove the support. :)

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei