bambulab/BambuStudio #1162: Build plate cannot be send to the printer (Exception:std:bad_alloc)

ekatus, 2023-01-24:
**Describe the bug**
Error upon sending build plate to the P1P printer - 

**To Reproduce**
Steps to reproduce the behavior:
1. slice model
2. print plate
3. confirm pop up and hit send
4. See error

**Expected behavior**
send 3mf to the cloud and initiate print

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - Ubuntu 22.04 on Intel Nuc 8 i5 16GB

lanewei120, 2023-01-24:
> **Describe the bug** Error upon sending build plate to the P1P printer -
>
> **To Reproduce** Steps to reproduce the behavior:
>
> 1. slice model
> 2. print plate
> 3. confirm pop up and hit send
> 4. See [...]

@ekatus 

thanks for reporting this  issue

it should be the same with https://github.com/bambulab/BambuStudio/issues/760#issuecomment-1397374901
could you check the details in that issue for the solution?

lanewei120, 2023-03-01:
already in v1.5 just released

closed 2023-03-01 by lanewei120