bambulab/BambuStudio #1166: Statistics for all the plates

OmiCron07, 2023-01-24:
We have the statistics one plate at a time and we need to manually add all the numbers to get the total print time, total plastic weight, total waste and so on.

If we have a large number of plates, this process is tedious. Could we have the statistics for all the plates?

SaltWei, 2023-03-01:
[image]
Add into v1.5.0 Public beta version

Hologos, 2023-03-04:
@SaltWei Can we reopen this FR and add the totals for model, flushed and totals like we get on the plate statistics? We get only totals per color.

[image]

SaltWei, 2023-03-07:
Accept. We will improve it
@LiZ-Li-BBL Please see this suggestion

Hologos, 2023-09-01:
@LiZ-Li-BBL Bump :-)