bambulab/BambuStudio #1167: Purge tower weight

OmiCron07, 2023-01-24:
The purge tower weight is not included anywhere, not even in the print cost. The statistic should indicate the purge weight, either in the flushed filaments or separately.

Left with purge tower / Right without
[image]

MadMax389, 2023-01-26:
Interesting. For me, the flushed weight stays the same and it appears the tower is included in the model weight. I always thought the printer would flush into the prime tower so that the flushed volume stays the same, just that some of the flush wound up in the tower. Doesn't seem to be the case in either example.
[image]

SaltWei, 2023-08-10:
@XunZhangBambu Please move the wipe tower filament consumption from  model to flushed. Now it's calculated into model. This is wrong.

XunZhangBambu, 2023-11-14:
In 1.8 beta , wipe tower filament consumption has been seperated fropm model .
[image]

closed 2023-12-21 by QingZhangBambu