bambulab/BambuStudio #1168: Video Quality in Bambu Studio

CCS86, 2023-01-24:
I posted this issue on the user forum 2 weeks ago and have not gotten a response:

https://forum.bambulab.com/t/video-quality-bambu-studio-vs-bambu-handy/3135

Basically the video in Bambu Studio is far too dark and has very choppy motion. More details, photos and videos in that thread.

bambu123, 2023-01-29:
We use your system's direct-show video decoder, which may be different from others.
Some Media software may also provide other decoders.

CCS86, 2023-01-30:
> We use your system's direct-show video decoder, which may be different from others. Some Media software may also provide other decoders.

What is the solution then? There is a very big difference between BS and BH.

I even tried putting black vinyl along the back wall to help the camera auto exposure brighten the image. Still, Bambu Studio is too dark:

[image]

CCS86, 2023-01-30:
Once it starts printing I can't see much at all:

[image]

bambu123, 2023-01-31:
@CCS86 
If you like, please contact me with discord, my account is chunmao.guo#3806.
I need to debug something with your help.

bambu123, 2023-02-02:
After some debugging with the help of @CCS86, we got the same result with the same URL.
The mobile phone may have a video effect filter supplied by the system media player, making the video sharper and brighter.
Currently, we have no custom video effects in the studio or handy.
So to fix this issue, I suggest you guys give your printers a brighter environment.

closed 2023-03-02 by SaltWei

CCS86, 2023-03-02:
I'm not sure how this is complete. 

The light was already on inside the printer. The glass is tinted, so environmental light has almost no impact.

I really think you guys should consider finding a way to increase brightness or gamma in BS.

Plus, there is the issue of video stuttering in BS, which is not present in BH. These are both ongoing issues.

mdeeter, 2024-05-23:
I agree that this is an issue.    The printer camera really needs a manual exposure setting.

CCS86, 2024-05-23:
Even with additional interior lights added, exposure control just hammers it dark. 

Here is the view from outside the printer:

[image]

And here is what the camera sees in the same moment:

[image]

Histogram of the first image:

[image]

And the second one:

[image]

Even with a black build plate and the interior walls blacked out, trying to get auto-exposure to push the brightness up:

[image]