bambulab/BambuStudio #1171: All printer profiles and filament profiles are missing when upgrading BS to new versions

iandouglas, 2023-01-25:
**Describe the bug**
I'm honestly not sure if this is a bug, or intended behavior by Bambu Labs. But the last several versions on Windows, when I've upgraded from a previous version, all of my custom filament profiles, and any printer profiles with gcode changes (eg, the community gcode from Facebook) are all missing. Having to go back and rebuild that every time makes me want to never upgrade Bambu Studio in the future.

**To Reproduce**
Run an older copy of the software on Windows
Copy a printer profile to another name, change something like the start/end gcode
Do the same for some filaments.
Upgrade to a newer version
Printer profile is missing, filament profiles are missing.

**Expected behavior**
As a user, I'd like to maintain my previous profiles when upgrading the Bambu Studio softwware.

If something changes in Bambu Studio where my custom profiles are somehow obsolete or otherwise rendered unusable, even then, allow me to select the old one and make changes or provide information on WHY the profile is inactive or direct me on which setting(s) are affecting it so I don't have to remember this stuff or make frequent manual backups of everything.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
- latest public release of Bambu Studio, but has happened the previous 2 updates as well

wirlybird, 2023-01-25:
All you have to do is log out of Studio and log back in.  Look on the FB groups.  This gets asked many times.

iandouglas, 2023-01-25:
We shouldn't have to do this though. As a user, I want to just upgrade the
software and leave my custom profiles alone.

iandouglas, 2023-01-25:
> Your profiles are stored on the Bambu cloud not your computer. The upgrade
> has no idea about your profiles until you log out and back in.

As a user, I should not have to log out to synchronize this.

lanewei120, 2023-01-28:
thanks for reporting this issue

it is indeed not a good behavior,
we will refine this logic in next release

closed 2023-03-01 by SaltWei

nicwilson58, 2023-03-03:
You do not get away with closing this bug so easily,  I have logged out and in a dozen times,  since the updated version and all filament profiles are gone,  and the AMS sync button is also missing.

SaltWei, 2023-03-03:
Could you share the log file under this path? [image] It's very easy to reopen and keep going