bambulab/BambuStudio #1172: [*Feature Request*] Pause fans when the current print is paused

BambuLabDiscord, 2023-01-25:
Currently, when pausing a print from Bambu Studio or from the device screen, the fan speed is not adjusted. A user can manually turn off the fans, but will then need to reset the fans to their prior speed after resuming the print.

This feature request is to turn off the fans when a print is paused (including the noisy mainboard fan, if possible) and return them to the prior speed when the print is resumed.

julie777, 2023-02-22:
I was thinking about this also. I wondered if the hot end fan needed to remain on while the hot end cools to some temperature. 

Currently it appears to me that I can set a time delay and after that time the fan will turn off, which doesn't seem like the best way. 

It also appears that the time delay (default 15 minutes) is also used to turn off the bed heater. I would think that it should turn off as soon as the print completes to allow the bed to cool so the part can be removed easier.

If I am misunderstanding, please let me know.

mattvenn, 2024-08-06:
I would also like to see this!

CallmeBigfat, 2025-06-15:
+1 the H2D does this and its amazing! the side fan is off during pause

tomasz90, 2025-08-21:
Can bambu implement this????