bambulab/BambuStudio #1174: [*Feature Request*] Save new K settings to the filament profile so K value will change when using different filament.

SeaStorm00, 2023-01-25:
This will allows user to easily change filament material without have to enter the correct K value manually.  This will prevent users from using the wrong K value when switching material.

gabrielf-tci, 2023-01-26:
Agree! The K value doesn't make a whole lot of sense being in the AMS section and stored into firmware NVRAM, when it is filament specific. As it is you would need to keep filament profiles for each filament and then additionally keep a separate log of K value to filament. Not very intuitive

gabrielf-tci, 2023-01-26:
Note that this could be deployed in two stages,

Stage 1, Slicer only "K" value storage under the Filament section, all filament profiles default to NULL "K" value, which will then pull K value from firmware, same as current functionality. If user ticks a checkbox to specify "K" value in filament section, slicer will push updated "K" value to relevant AMS Slot / Ext. Spool as needed, when sending print to printer via cloud.

Stage 2, firmware update on printer side to open up some NV or SD storage to sync filament profiles from slicer to AMS/ext spool. then, when that particular filament type is chosen from AMS tab, it would auto populate "K" value from saved value. That way filament "K" value can be stored in print history for example, and set proper value upon re-print.

p0thi, 2023-05-30:
I would like that too. 👍 It is really annoying if one forget to change the K value after changing the filament.