bambulab/BambuStudio #1177: Spiral Z-Hops and "Avoid Crossing Wall" can combine to make badly ordered hops, causing wall collisions

psiberfunk, 2023-01-25:
**Describe the bug**
In some situations, Using spiral Z-Hops and "Avoid Crossing Wall" (ON) can combine to make badly ordered hops, causing wall collisions.  In these situations the spiral retract happens AFTER the travel move instead of before it!  This negates the whole purpose of the Z-hop and causes significant surface artifacts on the print.

If "Avoid crossing wall" is turned OFF, the planning bug disappears. Alternatively, if normal z-hops are used instead of spiral (but "Avoid crossing wall" is left on) the bug also disappears.  It is only when BOTH are on that the bug seems to happen in this example. 

**3mf File for This Bug**
Pre-sliced and unsliced files are provided here:
[3MF_Files_AvoidCrossingWallsSpiralZhopGlitch.zip](https://github.com/SoftFever/BambuStudio-SoftFever/files/10501984/3MF_Files_AvoidCrossingWallsSpiralZhopGlitch.zip)

**To Reproduce**
Scanning through the GCODE (see movie attached here) shows the problem is that the order of operations is wrong:
1)Wipe
2)Travel move to desired location
3)Spiral hop (retract)
4)Unretract    [in the same spot as the retraction just happened!]

This is incorrect,   the right ordering of operations  is   

1)Wipe
2)Spiral hop (retract)
3)Travel move to desired location
4)Unretract    

**Movie showing bad order**
https://user-images.githubusercontent.com/19777571/214640987-78d643fc-e470-4ef5-ba5a-d26c8c551224.mp4

**Desktop (please complete the following information):**
 - MacOS
 - V1.4.2

**Cosmetic impact that this causes:**
With the issue:
[image]

Without the issue (by turning off avoid-crossing-walls):
[image]

psiberfunk, 2023-01-26:
P.S.  I don't know what's causing the issue with the pock marked surface on the ears here, but it's clearly not this bug and appears to be a separate retraction or other related issue.

npoulos, 2023-01-27:
Found the exact same issue!

Retracts > wipes > travels > arrives to start next line > z hop spirals > unretracts

Couldn't figure out what combination of settings did it. Looks like Psiberfunk figured it out!

psiberfunk, 2023-01-27:
I've printed like 20 of these things , a whole army just to track this bug down.

psiberfunk, 2023-02-07:
Note: this got fixed in the SoftFever fork here: https://github.com/SoftFever/BambuStudio-SoftFever/pull/279  Maybe pull that fix in ?

QingZhangBambu, 2023-02-08:
We've fixed it. just release it later

psiberfunk, 2023-02-08:
> We've fixed it. just release it later

lol :    Don't worry, the thing you can't see is real :)

closed 2023-03-02 by SaltWei