bambulab/BambuStudio #1183: Renaming the file when "sending to printer SD card" does not work.

threepoint2, 2023-01-26:
**Describe the bug**
Renaming the file when "sending to printer SD card" does not work. The file is sent with the project name and not the edited name

**3mf File for This Bug**
Not related to 3mf

**To Reproduce**
1. Slice file
2. Send plate
3. On the "Send to Printer SD card screen" click the pencil icon to change the name of the file
4. Click send.

**Expected behavior**
When viewing the file on the printer, it should have the changed name but it still has the original project name/file name.

**Screenshots**
This doesn't need a screen shot. The file name chosen on the Send to Printer SD card screen does not work.

**Desktop (please complete the following information):**
Windows 11
BS 1.4.2.13

threepoint2, 2023-01-26:
Solved.

closed 2023-01-26 by threepoint2

Guilouz, 2023-02-05:
@threepoint2 How do you fix that ? I have same issue.

aliasmeum, 2023-02-22:
I am afraid this is not solved, and that it is more serious than the “Rename file while sending” function not working. When using 3mf-files from Fusion360 it seems the name of the original part in Fusion360 is stuck no matter what you call the file. When you change filename, or change something in a project, and then send it to the printer you have no way of knowing what file is actually on the SD-card because it names the file the same as before the change. This behavior leaves the “Send to printer” function untrustworthy and thereby useless in the current state. I am really happy that Bambulab added this functionality, since the “Send to printer” function is excellent to use for files that are printed repeatedly, and I hope they fix this as soon as possible.

This is what I have done to try to reproduce the issue.

Fusion360:
-Save “A v1” as mesh in Fusion > “A.3mf”
-Save “B v2” as mesh in Fusion > “B.3mf”

Windows10 File explorer:
-Rename file “A.3mf” > “A-Fusion.3mf”
-Rename file “B.3mf” > “B-Fusion.3mf”

Bambu Studio 1.4.2.13:
-New project
-Add > “A-Fusion.3mf”
-Save project as > “A-Studio.3mf”
-Slice
-Send
Result: File sent to printer is named “A v1”

[image]

[image]

[image]

-New project
-Add > “B-Fusion.3mf”
-Save project as > “B-Studio.3mf”
-Slice
-Send
Result: File sent to printer is named “B v2”

[image]

[image]

[image]

-Open project > “A-Studio.3mf”
-Delete “A-Fusion.3mf”
-Add “B-Fusion.3mf”
-Save project as > “AtoB-Studio.3mf”
-Slice
-Send
Result: File sent to printer is named “B v2”

[image]

[image]

[image]

[image]

lestie198, 2025-06-05:
I am still having this issue in 2.1.0.59 on Windows 11.