bambulab/BambuStudio #1184: Please allow AMS+printer to work with TPU

AbeFM, 2023-01-26:
**Describe the bug**
Printer refuses to print with TPU and AMS, even though I've never had an issue with many rolls of TPU

**To Reproduce**
Load TPU into AMS
Enter that information into AMS
Try to print

**Expected behavior**
Printer would allow you to print

I end up making fake profiles, called things like "TPU but faked as PLA" and then I have to remember what's really in the printer versus what I told it was in the printer to get around the restriction.

The other day I got a VERY nasty jam because the printer was printing TPU as thought it was PLA.

If TPU cannot be used in the AMS, do not list it as a choice, and have the AMS complain when you load it. Allowing materials the printer refuses to hit the right temperature for will very predictably lead to disasters.

Asking users to create fake profiles and forcing them to remember all the work-arounds necessarily means valid safety checks will not be performed. Telling your support staff to ask people to do this risks damage to the machine which Bambu should be responsible for.

It would be MUCH better to have an error pop up and advise you not to print, but allowing it, so profiles will match the capabilities of the machine.

I would STRONG recommend creating a "TPU > 92A" and a "TPU <92A" menu item on the AMS, and when someone selects TPU <92A, display a warning and refuse to use it, and/or give people an option to acknowledge the risk and do it anyways.

Or at least please respond to the topic and give some feedback on how to use the machine within the framework. As an engineer, it's bad practice to encourage users to have a bunch of fake profiles, work arounds, etc. The risk of damage to the machine is much higher with the current system.

SaltWei, 2023-01-28:
Response from printer/AMS firmware team:
This is limited by hardware rather than a bug of slicer software.
The AMS instructions indicate that TPU is not supported.
[image]
We do have the option of TPU for the third material configuration of AMS at present, and this is unreasonable and we will remove in the later firmware.
And for some risky materials, we will add a hint to the user's selection.

DzzD, 2023-02-05:
Would be awesome to print TPU with AMS

The AMS unload & reload of the filament should probably need to be done more slowly for TPU. Is this possible via some gcode or settings ?

AbeFM, 2023-02-05:
Even if not used for MM printing, just for load/unload would be awesome - especially so it worked so well in the past.

DzzD, 2023-02-18:
So, finally would it be possible to have a precise desvription of what the bad things that can happen with TPU ? I believe everything after the "buffer" works as TPU can be printed without AMS, so does problems come from the buffer ? the AMS itself ?

Danne980, 2023-02-23:
Is there a practical reason with all these Y splitters that Bambu Labs can´t update software to use external spool in combination with AMS. I am looking at buying a Qidi tech  i-fast printer just to be able to combine TPU with other filament. Otherwise an updated AMS that works with TPU would ofcourse be even better.

AbeFM, 2023-02-23:
The good news is it does work, it's just very hard to set up fake profiles
(like I've been told to run by Bambu for PET-g and basically anything else
where automatic protections keep you from doing what works).

Instead of being straightforward about where the limitations are, Bambu is
sticking to a claim on this one that experience does not agree with.

Just the other day I jammed up my extruder when the machine auto selected
"pla" for a print that was actually tpu. I am curious what these issues
are, I've had more failed prints with PET-g on stock settings than I ever
did on tpu with stock settings, until they took TPU away.

I would sure appreciate not needing to buy a second printer or hack parts
into this one just to make it do what it demonstrably can do without issue.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

DzzD, 2023-04-20:
I have tried many times without success, the weak seems not to come from the feeder or the hub but rather from the output of the AMS and its motor

AbeFM, 2023-04-20:
Which tpu are you using?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

DzzD, 2023-04-21:
> Which tpu are you using? Thanks! -Abe. Sent from my "smart"phone, please excuse brevity and Swype-oes
> […](#)

Well I finally succeed using a different slot on the AMS :)

ESun TPU 95A Green : 

[image]

I will now try multi-material with TPU.

I think there are still some works to make it more reliable, the motor seems to works more on this spool than other.

sodachen22, 2023-04-21:
This is an interesting attempt, but there is no official guarantee that TPU will work on AMS.

DzzD, 2023-04-22:
> This is an interesting attempt, but there is no official guarantee that TPU will work on AMS.

Sure and for now it is absolutly not reliable, TPU really dont like to be push/pull multiple times...

DzzD, 2023-04-22:
Really hope that one day, you will give us access to the firmware as an open source project, I would love to address the TPU "problem" as well as the filament change speed.

For instance when printing from two AMS, it would be good in some cases  to not retract the filament back to the AMS, but only back just before the input of the HUB.

AbeFM, 2023-04-22:
> This is an interesting attempt, but there is no official guarantee that TPU will work on AMS.

Yeah, like I was saying, I've used it so much, I don't see what the big deal is. I've never heard of anyone having a problem.

> Well I finally succeed using a different slot on the AMS :)

So the slot that didn't work remains not working? Which slot is it? I wonder if there's a tube not fully seated or some such in your machine, or if the tip on your TPU hadn't been square, etc.

And I would REALLY appreciate if Bambu could be more explicit with why they feel TPU doesn't work.

sodachen22, 2023-04-23:
TPU material is too soft and easily stuck in AMS. Please take the two wiki pages as reference https://wiki.bambulab.com/en/knowledge-sharing/notes-AMS
https://wiki.bambulab.com/en/x1/manual/compatibility-and-parameter-settings-of-filaments

AbeFM, 2023-04-23:
In practice, it is not. There are grades of tpu, the TPUs with a "Shore
Hardness" of 95A work well in the AMS, in my experience.

Even if used with limited changes (one or two two layer changes) or just to
keep the material in the machine for single material tpu prints, it's VERY
useful to support it.

Please reconsider supporting it, especially in a limited changes-per-print
capacity. I would be happy to beta test anything necessary to make it work.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

shin-kamui, 2023-05-06:
Its hit or miss.  I agree that they should allow users to override the settings and print in the AMS with a warning in the slicer instead of blocking it entirely and forcing people to alter other materials to bypass the restriction.  I think with the latest update adding the beginnings of external spool management to the slicer and printer, it would be beneficial to add support for automating external spool + AMS printing.  This would enable some of the much softer TPUs and Flexible PLAs to be combine into prints as well.

AbeFM, 2024-07-10:
The issue is that, by lying to the machine, you get fast load/unloads. You miss the opportunity to synchronize motors. In short, you take something that had a 95% reliability and turn it into something with 60% and higher consequences for failure than you would if you just left it to expert users willing to take the risk.

gaolegao-lx, 2024-07-11:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

AbeFM, 2024-08-08:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

Is this for me? I believe I'm asking for a feature change, but if starting/continuing a support ticket will move this forward, I most certainly will.

josser, 2024-08-19:
Even in official TPU Printing guide it said that you can print TPU from AMS harder than 55D https://wiki.bambulab.com/en/knowledge-sharing/tpu-printing-guide 

But how to do that?

rogery555, 2024-08-22:
This is so frustrating , sure keep the warnings but at least give us an option to override to run harder TPU profiles (ideally with slower AMS motor speeds).  Even if something goes wrong there is nothing catastrophic about the failure so why is there such a restriction.?