bambulab/BambuStudio #1185: Every Model In The Project Is Mysteriously Duplicated And Superimposed

MadMax389, 2023-01-26:
**Describe the bug**
I have a project with 8 plates comprising about 20 models. I don't know when exactly the problem happened, but I had been opening and closing the project for about a week. I started a print on one of the plates and after about 10 layers noticed that the model was rough and the nozzle was scraping the surface. I stopped the print and tested the printer on a cube primitive with no problem. I then noticed that there were two models in the slicer exactly superimposed on top of each other, which is why the printer was double-depositing filament. Later I realized that every model in the project was duplicated and superimposed. I'm sure I didn't clone them by accident since I don't think you can clone multiple models and the clone is never exactly superimposed. Really weird. Posted here mostly to see if anyone else has experienced this.

**3mf File for This Bug**
Since the model is not open-licensed, I have included a .3mf with a subset of the models on 2 plates. If needed I can email the whole project to Bambu instead of posting here.
[~BS A10 Duplicates 2 Plates.zip](https://github.com/bambulab/BambuStudio/files/10513437/BS.A10.Duplicates.2.Plates.zip)

**To Reproduce**
Unable to reproduce. One-off weird issue, hasn't happened before or since.

**Expected behavior**
Models don't duplicate themselves.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win 11
 - Version: 1.4.2.13

paulhdietz, 2023-01-27:
I experienced the same issue. Printed as if the model was duplicated on itself. Upgraded to latest version of Studio, and now it slices correctly, so can not reproduce the bug.

SaltWei, 2023-01-28:
We are trying to reproduce this bug. But really hard to reproduce.
Other 2 users met same issue as well and reported issue to us by app.

MadMax389, 2023-01-28:
I haven't seen it in any other project. The only thing I can think of is that I sometimes leave the project open and then my computer goes to sleep after a few hours. It's typically not a problem when the computer wakes up, BUT if the device page is open with the camera playing in the project however, the computer never goes to sleep. I remember one morning that I had a Windows update in the night sometime and the computer had rebooted and restored all the open windows after the update.  I don't remember if the project was open or not, I can't pinpoint it to this event. Is this happening in MacOS as well as Windows?

SaltWei, 2023-01-29:
> I remember one morning that I had a Windows update in the night sometime and the computer had rebooted and restored all the open windows after the update. I don't remember if the project was open or not, so I can't pinpoint it to this event.

[image]
Did you see this pop-up window after these operations？

lanewei120, 2023-01-29:
@MadMax389 
Could you zip all the log files in C:\Users\user.name\AppData\Roaming\BambuStudio\log and upload here?

we will have a check

MadMax389, 2023-01-29:
> > I remember one morning that I had a Windows update in the night sometime and the computer had rebooted and restored all the open windows after the update. I don't remember if the project was open or not, so I can't pinpoint it to this event.
> 
> [image] Did you see this pop-up window after these operations？ Or did you use ctrl+Z to do any undo/redo operation?

Sorry, i don't recall seeing that dialog. I do use ctl-z when preparing prints, but haven't noticed any ill effects. I had responded to this FB post on Jan 26 when I noticed the the user had an issue that looked exactly like my print problem. I noticed my problem a day or two earlier (maybe Jan 24?). I opened the ticket after I saw his post. https://www.facebook.com/groups/bambulabofficialusers/posts/942968376647182/?comment_id=943124329964920&reply_comment_id=943217426622277&notif_id=1674785226235918&notif_t=group_comment&ref=notif 

Here are my log files...
[debug_network_Sat_Jan_14_07_59_16.zip](https://github.com/bambulab/BambuStudio/files/10529545/debug_network_Sat_Jan_14_07_59_16.zip)

lanewei120, 2023-01-30:
we found the root cause, 
In some corner cases, opening project from "recent projects" was blocked, and when you opened again,
the previous opening will cause the model loaded twice

we will fix this issue in next version

thanks for reporting this issue

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei