bambulab/BambuStudio #1186: Filament inventory tracking

nkamenar, 2023-01-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Tracking filament inventory for 3D printing is a bit of a chore and I think there is a lot of opportunity for automation and some really nice quality of life features that no other slicing software has right now.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I would like to be able to enter my filament inventory into a section of Bambu Studio
For each entered filament I should be able to enter the following (* = required):

- Filament type (PLA, PETG, etc) *
- Brand *
- Color *
- Diameter *
- Filament profile (selectable from existing filament profiles) *
- Quantity (remaining grams) *
- Notes (for misc. info like maybe which storage container you put it in)

Then when getting ready to 3D print you can either select a filament profile for each AMS slot like you do in bambu studio currently, or you could select from your existing inventory. Using a filament from your inventory will automatically apply the appropriate profile for that filament and decrement the quantity. Decrementing the quantity would ideally happen in real time based on actual progress of the print but I am not sure the X1 has those sensors. Worst case when you click print it would just subtract out the quantity of the full print from the filament and if you stopped the print mid way or something you would have to weigh the spool and update manually.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Octoprint I believe has plugins to sort of achieve some of this but it's not nearly as full featured as what I described above.
Alternatively many people just track inventories manually in an excel spreadsheet and sure that works but it's manual and tedious.

**Additional context**
Add any other context or screenshots about the feature request here.
I think adding this functionality would really give Bambu Studio an edge and would be a killer feature that would really attract a lot of prospective hobbyists into the Bambu ecosystem.

Allram, 2023-01-27:
This is an excellent suggestion! +1

Popoboxxo, 2023-04-01:
Would be awesome!

Beastly-Kitty, 2023-07-04:
I would like to see something integrated into the software to track filament. I have been using Octoprint's PrintJobHistory for years and would love to be able to regularly pull the data from my print jobs and see what filaments I use most over the years. It also helps me keep track of my total filament expenditure over time. Ideal for hobbyists / small business owners!

arcreigh, 2023-08-26:
+1 to this being able to keep an inventory of available filament would be awesome.
But also with the addition of price per spool. (so we can have accurate price per print data)

kennethfinn, 2023-10-07:
I support this 100%  👍

Rufics, 2024-02-11:
+

SaltWei, 2024-02-19:
@nkamenar Thank you very much! We will evaluate your suggestion carefully.

higgysmalls-ky, 2024-10-10:
+1 as this is a much needed feature that many other apps already have available.

jjvelar, 2024-11-04:
Hi,
That would be of great value for AMS Lite too.
Is this in the pipeline?
Thanks!

braedonwatkins, 2024-12-02:
@lanewei120 @SaltWei are you guys actively working this? It's my first feature here but I've got experience I think I might be able to tackle it.

jjvelar, 2024-12-02:
would be great to have this for AMS Lite.

MartinChan3, 2024-12-03:
@higgysmalls-ky  Can you offer an example? Maybe this will help developer to know more about what they should do.

FaberO42, 2025-01-01:
Big +1 for something along these lines.
Bambu great for managing 3D printing process, but significant room for improvement in managing key (filament) consumables use.
Also requires to support more than just Bambu brand filaments.

glassoboe, 2026-09-26:
Most of this now exists in the Filament Manager, but the automatic part still doesn't. Suggestion: store each spool's remaining weight against its RFID tag UID, deduct usage after each print, and refine the number when the AMS re-estimates on reload. The AMS would then recognize a spool when it goes back in rather than starting from scratch, and the inventory would stay current without selecting spools or weighing them. For third-party spools, letting users pair their own NFC sticker with an inventory entry would extend the same tracking to everything on the shelf.