bambulab/BambuStudio #1187: Plate check happens too late

npoulos, 2023-01-27:
**Describe the bug**
When the printer checks if the correct plate is installed it does it as the last step before printing calibration lines. This means homing, heating and bed levelling happens before it checks the plate. First, it's a huge waste of time but also if you've forgotten to actually add the plate it damages the bed.

**Expected behavior**
It should home first, then check plate (and alert if needed), then level/calibrate.

There should also be an option to turn off the plate-check feature as a lot of people are using aftermarket plates. Having to manually tell the printer to continue printing after it says the wrong plate is on is bad UX. Let me tell the printer that i'm not using one of the Bambulab plates first.

Merlwynd, 2023-01-27:
AGREE!!

SaltWei, 2023-01-28:
Thank you very much!
Has transferred this issue to printer team from which start gcode is provided.

SaltWei, 2023-01-29:
Response from printer team:
Has been moved before bed level. Please wait for next update