bambulab/BambuStudio #1188: Allow multiple values of volumetric speed based on nozzle size

npoulos, 2023-01-27:
**Is your feature request related to a problem? Please describe.**
Currently a filament profile allows you to specify a max volumetric speed. The default Bambu ones have a different speed for the 0.2 nozzle compared to the other nozzles.

The problem here comes when creating a custom filament profile. You have to make a filament profile for the different nozzle sizes because the volumetric speed between nozzles are different.

**Describe the solution you'd like**
allow an input for max volumetric speed for

0.2
0.4
0.6
0.8

size nozzles.

This way you don't have to end up making 4 different profiles to change one value.

RobinWhitfield, 2023-02-15:
The issue with this idea is that different sized nozzles also require other settings to be changed (mainly cooling) to get a good print

SaltWei, 2023-03-02:
> This way you don't have to end up making 4 different profiles to change one value.

As you said, they are different profiles and different json file. A lot of difference including cooling settings.
This is huge change to merge them together.