bambulab/BambuStudio #1189: Linux: Add camera support for P1P

spuiuk, 2023-01-27:
PR #1045 adds camera support for X1 series of printers. This however doesn't work on a P1P [since the stream on the P1P uses mjpeg stream instead of H264 for the X1 series.](https://github.com/bambulab/BambuStudio/pull/1045#issuecomment-1398127318)

[A change was proposed in that PR which would also make it work for the P1P. ](https://github.com/bambulab/BambuStudio/pull/1045#issuecomment-1398180654). I have gone ahead and[ tested the proposed change and can confirm that it works](https://github.com/bambulab/BambuStudio/pull/1045#issuecomment-1405735534) for the P1P.

The patch I used  on the latest v1.4.2 branch as mentioned in a comment to PR #1045 is as follows
```
diff --git a/src/slic3r/GUI/Printer/gstbambusrc.c b/src/slic3r/GUI/Printer/gstbambusrc.c
index bfde2f57..5f0f9cae 100644
--- a/src/slic3r/GUI/Printer/gstbambusrc.c
+++ b/src/slic3r/GUI/Printer/gstbambusrc.c
@@ -78,7 +78,7 @@ GST_DEBUG_CATEGORY_STATIC (gst_bambusrc_debug);
 static GstStaticPadTemplate srctemplate = GST_STATIC_PAD_TEMPLATE ("src",
     GST_PAD_SRC,
     GST_PAD_ALWAYS,
-    GST_STATIC_CAPS("video/x-h264,framerate=0/1,parsed=(boolean)false,stream-format=(string)byte-stream"));
+    GST_STATIC_CAPS_ANY);
 
 enum
 {

```
My repo with this patch is at https://github.com/spuiuk/BambuStudio/tree/test_pip_video
and a test Appimage available at [www.spui.uk/BambuStudio/BambuStudio_F37_v01.04.02.13-5-g5bd2b114.AppImage](https://github.com/bambulab/BambuStudio/pull/www.spui.uk/BambuStudio/BambuStudio_F37_v01.04.02.13-5-g5bd2b114.AppImage)
(The Appimage is built slightly differently for my Fedora 37 system)

I have been unable to test if this still works with the X1 series as I do not have the hardware with me.

lanewei120, 2023-01-28:
thanks for doing this~

we already verified here, it works on both X1 series and P1P
already submitted this patch to branch v1.4.2 at https://github.com/bambulab/BambuStudio/commit/9f323d419700bcbde811a038442af7da136073f1

lanewei120, 2023-01-29:
@spuiuk 
I  just found the video from P1P only has the first frame
it is not updated

is this the same on your side?

spuiuk, 2023-01-29:
No. The camera works fine. It is a low number of frames as expected for the P1P. I'll try rebuilding for the latest v1.4.2 branch and retry.

lanewei120, 2023-01-29:
I found the pts is wrong on my side
I will submit a patch to fix it soon.

anyway, it is strange it works well on your side

spuiuk, 2023-01-29:
Could it be because some versions of libraries could be different because of the Fedora build? I've uploaded a screen cast. 
[Screencast from 2023-01-29 16-17-30.webm](https://user-images.githubusercontent.com/1053298/215339759-f404bcef-5875-42a7-a72e-65c9dc6af35e.webm)

lanewei120, 2023-01-30:
thanks

just submitted a patch to fix it on v1.4.2 branch
now it works well on my side also.

jwise, 2023-01-30:
Interesting.  I wonder if we should just do this for h.264 also.  It's not like we have a particularly steady frame rate anyway, and the two clocks could in theory diverge in such a way that it makes trouble later.

lanewei120, 2023-02-03:
oops
I found it always crash on my Ubuntu
and always successfully on Fedora

it seems related to the jpeg decoder:

0:00:21.986561795 754446 0x55d2c0e56de0 INFO      GST_PLUGIN_LOADING gstplugin.c:987:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/[libgsttypefindfunctions.so](http://libgsttypefindfunctions.so/)" loaded
0:00:21.986607582 754446 0x55d2c0e56de0 INFO               GST_EVENT gstevent.c:892:gst_event_new_caps: creating caps event image/jpeg, width=(int)1280, height=(int)720, sof-marker=(int)0
0:00:21.986617567 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstelement.c:1016:gst_element_get_static_pad: found pad typefind:src
0:00:21.986622430 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstelement.c:1016:gst_element_get_static_pad: found pad typefind:sink
0:00:21.986652499 754446 0x55d2c0e56de0 INFO                GST_PADS gstpad.c:2382:gst_pad_link_prepare: trying to link typefind:src and decodepad0:proxypad3
0:00:21.986658511 754446 0x55d2c0e56de0 INFO                GST_PADS gstpad.c:2590:gst_pad_link_full: linked typefind:src and decodepad0:proxypad3, successful
0:00:21.986660772 754446 0x55d2c0e56de0 INFO               GST_EVENT gstevent.c:1660:gst_event_new_reconfigure: creating reconfigure event
0:00:21.986905740 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstelement.c:1016:gst_element_get_static_pad: found pad audio-sink:sink
0:00:21.987387944 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstelement.c:1016:gst_element_get_static_pad: found pad video-sink:sink
0:00:21.993299470 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstpad.c:2137:gst_pad_unlink: unlinking typefind:src(0x55d2bca60ec0) and decodepad0:proxypad3(0x55d2bca573e0)
0:00:21.993309859 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstpad.c:2192:gst_pad_unlink: unlinked typefind:src and decodepad0:proxypad3
0:00:21.993315266 754446 0x55d2c0e56de0 INFO                GST_PADS gstpad.c:2382:gst_pad_link_prepare: trying to link typefind:src and decodepad0:proxypad3
0:00:21.993318657 754446 0x55d2c0e56de0 INFO                GST_PADS gstpad.c:2590:gst_pad_link_full: linked typefind:src and decodepad0:proxypad3, successful
0:00:21.993320896 754446 0x55d2c0e56de0 INFO               GST_EVENT gstevent.c:1660:gst_event_new_reconfigure: creating reconfigure event
0:00:21.993333644 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstelement.c:1016:gst_element_get_static_pad: found pad video-sink:sink
0:00:21.993343784 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstpad.c:2137:gst_pad_unlink: unlinking typefind:src(0x55d2bca60ec0) and decodepad0:proxypad3(0x55d2bca573e0)
0:00:21.993347618 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstpad.c:2192:gst_pad_unlink: unlinked typefind:src and decodepad0:proxypad3
0:00:21.993561008 754446 0x55d2c0e56de0 INFO      GST_PLUGIN_LOADING gstplugin.c:987:_priv_gst_plugin_load_file_for_registry: plugin "/usr/lib/x86_64-linux-gnu/gstreamer-1.0/[libgstjpeg.so](http://libgstjpeg.so/)" loaded
0:00:21.993566096 754446 0x55d2c0e56de0 INFO     GST_ELEMENT_FACTORY gstelementfactory.c:489:gst_element_factory_create_with_properties: creating element "jpegdec"
0:00:21.993619194 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstelement.c:759:gst_element_add_pad:<GstVideoDecoder@0x7f26302f93c0> adding pad 'sink'
0:00:21.993627015 754446 0x55d2c0e56de0 INFO        GST_ELEMENT_PADS gstelement.c:759:gst_element_add_pad:<GstVideoDecoder@0x7f26302f93c0> adding pad 'src'
0:00:21.993641757 754446 0x55d2c0e56de0 DEBUG                jpegdec gstjpegdec.c:258:gst_jpeg_dec_init: initializing
Segmentation fault (core dumped)

lanewei120, 2023-02-03:
I have no idea on the core stacks(core dump thread is 754446):
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `../install_dir/bin/bambu-studio'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
167	../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S: No such file or directory.
[Current thread is 1 (Thread 0x7f256bea9640 (LWP 755248))]
(gdb) bt
#0  ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
#1  0x038710eec0af7efc in  ()
(gdb) info threads
  Id   Target Id                          Frame 
* 1    Thread 0x7f256bea9640 (LWP 755248) ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
  2    Thread 0x7f262ca81640 (LWP 754470) __futex_abstimed_wait_common64 (private=671123888, cancel=true, abstime=0x7f262ca782d0, op=137, expected=0, futex_word=0x7f262ca78400) at ./nptl/futex-internal.c:57
  3    Thread 0x7f2627fff640 (LWP 754484) 0x00007f264cee1d7f in __GI___poll (fds=0x7f261801f7c0, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  4    Thread 0x7f2649156f80 (LWP 754446) 0x00007f264cee1d7f in __GI___poll (fds=0x55d2baaf4680, nfds=5, timeout=188) at ../sysdeps/unix/sysv/linux/poll.c:29
  5    Thread 0x7f262ffff640 (LWP 754464) 0x00007f264cee47ed in __GI___select (nfds=14, readfds=0x7f262fff6440, writefds=0x7f262fff64c0, exceptfds=0x0, timeout=0x7f262fff6430) at ../sysdeps/unix/sysv/linux/select.c:69
  6    Thread 0x7f25c2544640 (LWP 754508) __futex_abstimed_wait_common64 (private=-1409255040, cancel=true, abstime=0x7f25c253a1e0, op=137, expected=0, futex_word=0x55d2bbd879a8) at ./nptl/futex-internal.c:57
  7    Thread 0x7f25c9ffd640 (LWP 754485) 0x00007f264cee1d7f in __GI___poll (fds=0x7f261401c9e0, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  8    Thread 0x7f25c1542640 (LWP 754514) __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55d2bca765a8) at ./nptl/futex-internal.c:57
  9    Thread 0x7f25c1d43640 (LWP 754509) __futex_abstimed_wait_common64 (private=-1043094792, cancel=true, abstime=0x7f25c1d3a3d0, op=137, expected=0, futex_word=0x55d2ba963a28) at ./nptl/futex-internal.c:57
  10   Thread 0x7f2646f04640 (LWP 754449) 0x00007f264cee1d7f in __GI___poll (fds=0x7f2638017550, nfds=7, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  11   Thread 0x7f259ffff640 (LWP 754528) 0x00007f264cee1d7f in __GI___poll (fds=0x7f25980073b0, nfds=3, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  12   Thread 0x7f25b89fd640 (LWP 754526) 0x00007f264cee1d7f in __GI___poll (fds=0x7f25a8001260, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  13   Thread 0x7f25c3546640 (LWP 755221) 0x00007f264ceae868 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7f25c353d490, rem=0x7f25c353d480) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
  14   Thread 0x7f25b99ff640 (LWP 754523) __futex_abstimed_wait_common64 (private=-1163258224, cancel=true, abstime=0x7f25b99f64d0, op=137, expected=0, futex_word=0x55d2b9d50d5c <Slic3r::_BBS_Backup_Manager::get()::m+124>) at ./nptl/futex-internal.c:57
  15   Thread 0x7f259f7fe640 (LWP 754535) __futex_abstimed_wait_common64 (private=-1631822848, cancel=true, abstime=0x7f259f7f5560, op=393, expected=0, futex_word=0x7f262d810d68 <send_cond_store+40>) at ./nptl/futex-internal.c:57
  16   Thread 0x7f25c0d41640 (LWP 755220) 0x00007f264cee1d7f in __GI___poll (fds=0x7f25a4003650, nfds=3, timeout=10) at ../sysdeps/unix/sysv/linux/poll.c:29
  17   Thread 0x7f259effd640 (LWP 754536) 0x00007f264cee47ed in __GI___select (nfds=52, readfds=0x7f262d8ac520 <mod_s>, writefds=0x7f259eff4410, exceptfds=0x0, timeout=0x7f259eff4400) at ../sysdeps/unix/sysv/linux/select.c:69
  18   Thread 0x7f2626ffd640 (LWP 754482) __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x7f2626ff45c0, op=393, expected=0, futex_word=0x7f260c00e990) at ./nptl/futex-internal.c:57
  19   Thread 0x7f2644b7b640 (LWP 754465) 0x00007f264ceae868 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=req@entry=0x7f2644b72590, rem=rem@entry=0x0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
  20   Thread 0x7f26277fe640 (LWP 754483) __futex_abstimed_wait_common64 (private=201390640, cancel=true, abstime=0x7f26277f5290, op=393, expected=0, futex_word=0x7f262504c160) at ./nptl/futex-internal.c:57
  21   Thread 0x7f25b91fe640 (LWP 754525) 0x00007f264cee1d7f in __GI___poll (fds=0x7f25b0001150, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  22   Thread 0x7f2647842640 (LWP 754448) 0x00007f264cee1d7f in __GI___poll (fds=0x55d2baa91fe0, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  23   Thread 0x7f259dffb640 (LWP 754545) __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55d2bd3d7df0) at ./nptl/futex-internal.c:57
  24   Thread 0x7f2571f47640 (LWP 755163) 0x00007f264cee47ed in __GI___select (nfds=57, readfds=0x7f2571f3e510, writefds=0x0, exceptfds=0x0, timeout=0x7f2571f3e4d0) at ../sysdeps/unix/sysv/linux/select.c:69
(gdb) thread 4
[Switching to thread 4 (Thread 0x7f2649156f80 (LWP 754446))]
#0  0x00007f264cee1d7f in __GI___poll (fds=0x55d2baaf4680, nfds=5, timeout=188) at ../sysdeps/unix/sysv/linux/poll.c:29
[... cut ...]
#7  0x000055d2b763f577 in wxEntry(int&, wchar_t**) ()
#8  0x000055d2b49e8844 in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...) at /home/lanewei/codes/clean_bambu/bamboo_slicer/src/slic3r/GUI/GUI_Init.cpp:64
#9  0x000055d2b3958cec in Slic3r::CLI::run(int, char**) (this=0x7fff8e4dc710, argc=1, argv=0x7fff8e4dc948) at /home/lanewei/codes/clean_bambu/bamboo_slicer/src/BambuStudio.cpp:452
#10 0x000055d2b396f217 in main(int, char**) (argc=1, argv=0x7fff8e4dc948) at /home/lanewei/codes/clean_bambu/bamboo_slicer/src/BambuStudio.cpp:2625

lanewei120, 2023-02-03:
@jwise any ideas？

jwise, 2023-02-03:
That's strange.  I would be much more inclined to believe that the `__longjmp_chk()` was crashing than the `__GI__poll()`, especially with that stack trace... What happens if you run it live in the debugger rather than opening up a core file after it crashes?  Tracking down a longjmp crash could be quite difficult, though, because by the time we actually crash in a longjmp, the calling stack for the crash is already gone... it might be possible to use gdb's `record btrace` to get a better understanding of what happened leading up to the crash?

That said, I have found that with the size of the slicer built in debug mode (1.5GB+ of debug symbols!), `gdb` becomes *very* slow on my machine.  I usually prefer `gdb`, but for Bambu Studio, I have been using `lldb` with a lot more success.

I don't have a P1P to reproduce this on my own but if it comes down to it, I guess I don't actually need one of my own, I only need one to be paired to my account.  So if you have one on the US cloud somewhere we can figure out how to pair it to my account so I can reproduce it, I suppose.

lanewei120, 2023-02-03:
> That's strange. I would be much more inclined to believe that the `__longjmp_chk()` was crashing than the `__GI__poll()`, especially with that stack trace... What happens if you run it live in the [...]

yes, you are correct
the crash happens in the thread calling __longjmp_chk()
`Thread 708 "source:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff15353640 (LWP 790836)]
____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
167	../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S: No such file or directory.
(gdb) info threads
  Id   Target Id                                            Frame 
  1    Thread 0x7fffed497f80 (LWP 790069) "bambustu_main"   0x00007ffff1222d7f in __GI___poll (fds=0x55555c7e0a80, nfds=5, timeout=445) at ../sysdeps/unix/sysv/linux/poll.c:29
  3    Thread 0x7fffebb83640 (LWP 790073) "gmain"           0x00007ffff1222d7f in __GI___poll (fds=0x55555c4318c0, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  4    Thread 0x7fffeb245640 (LWP 790074) "gdbus"           0x00007ffff1222d7f in __GI___poll (fds=0x55555ca3fa20, nfds=7, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  14   Thread 0x7fffd3fff640 (LWP 790089) "iotc_conn_hdlr"  0x00007ffff12257ed in __GI___select (nfds=14, readfds=0x7fffd3ff6440, writefds=0x7fffd3ff64c0, exceptfds=0x0, timeout=0x7fffd3ff6430) at ../sysdeps/unix/sysv/linux/select.c:69
  15   Thread 0x7fffe8ebc640 (LWP 790090) "iotc_taskmgr"    0x00007ffff11ef868 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=req@entry=0x7fffe8eb3590, rem=rem@entry=0x0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
  16   Thread 0x7fffd10a4640 (LWP 790095) "bambustu_main"   __futex_abstimed_wait_common64 (private=-872300704, cancel=true, abstime=0x7fffd109b2d0, op=137, expected=0, futex_word=0x7fffd109b400) at ./nptl/futex-internal.c:57
  28   Thread 0x7fffc77fe640 (LWP 790107) "bambustu_main"   __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x7fffc77f55c0, op=393, expected=0, futex_word=0x7fffac00e994) at ./nptl/futex-internal.c:57
  29   Thread 0x7fffc7fff640 (LWP 790108) "PressureMonitor" __futex_abstimed_wait_common64 (private=-1409222096, cancel=true, abstime=0x7fffc7ff6290, op=393, expected=0, futex_word=0x7fffc504c160) at ./nptl/futex-internal.c:57
  30   Thread 0x7fffd0ca3640 (LWP 790109) "HashSaltStorage" 0x00007ffff1222d7f in __GI___poll (fds=0x7fffb401f7c0, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  31   Thread 0x7fffc4ffd640 (LWP 790110) "ebsiteDataStore" 0x00007ffff1222d7f in __GI___poll (fds=0x7fffbc01c9e0, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  40   Thread 0x7fff6bffd640 (LWP 790132) "bambustu_main"   __futex_abstimed_wait_common64 (private=1543534736, cancel=true, abstime=0x7fff6bff31e0, op=137, expected=0, futex_word=0x55555d72a248) at ./nptl/futex-internal.c:57
  41   Thread 0x7fff68dff640 (LWP 790133) "bambustu_main"   __futex_abstimed_wait_common64 (private=1759470328, cancel=true, abstime=0x7fff68df63d0, op=137, expected=0, futex_word=0x55555c3a8b28) at ./nptl/futex-internal.c:57
  46   Thread 0x7fff63fff640 (LWP 790138) "bambustu_main"   __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55555e417d58) at ./nptl/futex-internal.c:57
  54   Thread 0x7fff62ffd640 (LWP 790146) "bambustu_main"   __futex_abstimed_wait_common64 (private=1547868816, cancel=true, abstime=0x7fff62ff44d0, op=137, expected=0, futex_word=0x55555c07fd5c <Slic3r::_BBS_Backup_Manager::get()::m+124>) at ./nptl/futex-internal.c:57
  57   Thread 0x7fff627fc640 (LWP 790149) "ReceiveQueue"    0x00007ffff1222d7f in __GI___poll (fds=0x7fff58000e70, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  58   Thread 0x7fff61ffb640 (LWP 790150) "ReceiveQueue"    0x00007ffff1222d7f in __GI___poll (fds=0x7fff54001260, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  59   Thread 0x7fff617fa640 (LWP 790156) "threaded-ml"     0x00007ffff1222d7f in __GI___poll (fds=0x7fff440073b0, nfds=3, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  60   Thread 0x7fff60ff9640 (LWP 790162) "bambustu_main"   __futex_abstimed_wait_common64 (private=-1537450752, cancel=true, abstime=0x7fff60ff0560, op=393, expected=0, futex_word=0x7fffe861fd6c <send_cond_store+44>) at ./nptl/futex-internal.c:57
  61   Thread 0x7fff3bfff640 (LWP 790163) "bambustu_main"   0x00007ffff12257ed in __GI___select (nfds=51, readfds=0x7fffe86bb520 <mod_s>, writefds=0x7fff3bff6410, exceptfds=0x0, timeout=0x7fff3bff6400) at ../sysdeps/unix/sysv/linux/select.c:69
  69   Thread 0x7fff3a7fc640 (LWP 790173) "bambustu_main"   __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55555ed633d0) at ./nptl/futex-internal.c:57
  669  Thread 0x7fff1a5f2640 (LWP 790775) "bambustu_main"   0x00007ffff12257ed in __GI___select (nfds=56, readfds=0x7fff1a5e9510, writefds=0x0, exceptfds=0x0, timeout=0x7fff1a5e94d0) at ../sysdeps/unix/sysv/linux/select.c:69
  706  Thread 0x7fff637fe640 (LWP 790831) "threaded-ml"     0x00007ffff1222d7f in __GI___poll (fds=0x7fff50000df0, nfds=3, timeout=20) at ../sysdeps/unix/sysv/linux/poll.c:29
  707  Thread 0x7fffe96bd640 (LWP 790832) "bambustu_main"   0x00007ffff11ef868 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7fffe96b4490, rem=0x7fffe96b4480) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
* 708  Thread 0x7fff15353640 (LWP 790836) "source:src"      ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
(gdb) bt
#0  ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
#1  0x897715e85cc64b75 in  ()`

jwise, 2023-02-03:
Very dismaying.  Yes, I would try using `record btrace`.  Maybe do something like:

```
$ gdb ./bambu-studio
[... it takes forever ...]
(gdb) run
[... let bambu studio start up, get ready to reproduce the crash ...]
^C
(gdb) record btrace # get gdb ready to record
(gdb) record stop # make it stop -- this would record only on a previous thread
(gdb) target record-btrace # enable tracing on new threads
(gdb) cont
[... reproduce the crash ...]
(gdb) record function-call-history
[... go get lunch, this is VERY slow on my machine ...]
```

There might be other ways to use the processor trace functionality that are less slow.  This one probably works... eventually...

jwise, 2023-02-03:
Oh, good news.  It 'only' seemed to take three minutes to decode the instruction trace (on my puny i7-10510U).  And once it decoded it, it seemed like it was happy.  So only a cup of coffee, maybe not all of lunch.

```
^C
Thread 1 "bambustu_main" received signal SIGINT, Interrupt.
[Switching to Thread 0x7fffe5ec0a40 (LWP 222636)]
0x00007fffef8c6d7f in __GI___poll (fds=0x60400057e890, nfds=5, timeout=357) at ../sysdeps/unix/sysv/linux/poll.c:29
29	../sysdeps/unix/sysv/linux/poll.c: No such file or directory.
(gdb) thread 726
[Switching to thread 726 (Thread 0x7fff8485c640 (LWP 223668))]
#0  0x00007fffef893868 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=req@entry=0x7fff848527c0, rem=rem@entry=0x0)
    at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
78	../sysdeps/unix/sysv/linux/clock_nanosleep.c: No such file or directory.
(gdb) show record btrace pt buffer-size
The record/replay pt buffer size is 16384.
(gdb) info record
Active record target: record-btrace
Recording format: Intel Processor Trace.
Buffer size: 16kB.
Recorded 51298 instructions in 4501 functions (0 gaps) for thread 726 (Thread 0x7fff8485c640 (LWP 223668)).
(gdb) record function-call-history  
4492	avRecvFrameData2
4493	BambuTunnelTutk::read(Bambu_Sample*)
4494	BambuTunnel::read_sample(Bambu_Sample*)
4495	gst_bambusrc_create
4496	usleep@plt
4497	usleep
4498	__GI___nanosleep
4499	__GI___clock_nanosleep
4500	__GI___pthread_enable_asynccancel
4501	__GI___clock_nanosleep
```

If you want an instruction-by-instruction trace of what happened, try `(gdb) record instruction-history -`.

lanewei120, 2023-02-03:
found some decode error, maybe I need to use the debug library of gstreamer?

```
(gdb) info threads
  Id   Target Id                                            Frame 
  1    Thread 0x7fffed497f80 (LWP 825174) "bambustu_main"   0x00007ffff1222d7f in __GI___poll (fds=0x55555c7ee7a0, nfds=5, timeout=331) at ../sysdeps/unix/sysv/linux/poll.c:29
  3    Thread 0x7fffebb83640 (LWP 825178) "gmain"           0x00007ffff1222d7f in __GI___poll (fds=0x55555c41beb0, nfds=2, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  4    Thread 0x7fffeb245640 (LWP 825179) "gdbus"           0x00007ffff1222d7f in __GI___poll (fds=0x55555ca2cb90, nfds=6, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  14   Thread 0x7fffea6bf640 (LWP 825194) "iotc_conn_hdlr"  0x00007ffff12257ed in __GI___select (nfds=14, readfds=0x7fffea6b6440, writefds=0x7fffea6b64c0, exceptfds=0x0, timeout=0x7fffea6b6430) at ../sysdeps/unix/sysv/linux/select.c:69
  15   Thread 0x7fffe96bd640 (LWP 825195) "iotc_taskmgr"    0x00007ffff11ef868 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=req@entry=0x7fffe96b4590, rem=rem@entry=0x0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
  29   Thread 0x7fffd3fff640 (LWP 825213) "bambustu_main"   __futex_abstimed_wait_common64 (private=1744963008, cancel=true, abstime=0x7fffd3ff6320, op=137, expected=0, futex_word=0x7fffd3ff6450) at ./nptl/futex-internal.c:57
  41   Thread 0x7fffc7fff640 (LWP 825225) "PressureMonitor" __futex_abstimed_wait_common64 (private=-1082070480, cancel=true, abstime=0x7fffc7ff6290, op=393, expected=0, futex_word=0x7fffbe04c160) at ./nptl/futex-internal.c:57
  42   Thread 0x7fff7b7fc640 (LWP 825226) "HashSaltStorage" 0x00007ffff1222d7f in __GI___poll (fds=0x7fffc801d380, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  43   Thread 0x7fff7affb640 (LWP 825227) "ebsiteDataStore" 0x00007ffff1222d7f in __GI___poll (fds=0x7fffd8005330, nfds=1, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  52   Thread 0x7fff711ff640 (LWP 825249) "bambustu_main"   __futex_abstimed_wait_common64 (private=1610646176, cancel=true, abstime=0x7fff711f51e0, op=137, expected=0, futex_word=0x55555d73e760) at ./nptl/futex-internal.c:57
  53   Thread 0x7fff709fe640 (LWP 825250) "bambustu_main"   __futex_abstimed_wait_common64 (private=1889489656, cancel=true, abstime=0x7fff709f53d0, op=137, expected=0, futex_word=0x55555c391728) at ./nptl/futex-internal.c:57
  57   Thread 0x7fff67fff640 (LWP 825254) "bambustu_main"   __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55555e428c48) at ./nptl/futex-internal.c:57
  62   Thread 0x7fff66ffd640 (LWP 825259) "bambustu_main"   __futex_abstimed_wait_common64 (private=1547885712, cancel=true, abstime=0x7fff66ff44d0, op=137, expected=0, futex_word=0x55555c069d58 <Slic3r::_BBS_Backup_Manager::get()::m+120>) at ./nptl/futex-internal.c:57
  65   Thread 0x7fff657fa640 (LWP 825268) "threaded-ml"     0x00007ffff1222d7f in __GI___poll (fds=0x7fff480073b0, nfds=3, timeout=-1) at ../sysdeps/unix/sysv/linux/poll.c:29
  66   Thread 0x7fff64ff9640 (LWP 825273) "bambustu_main"   __futex_abstimed_wait_common64 (private=1293668608, cancel=true, abstime=0x7fff64ff0560, op=393, expected=0, futex_word=0x7fffd1810b8c <send_cond_store+44>) at ./nptl/futex-internal.c:57
  67   Thread 0x7fff3ffff640 (LWP 825274) "bambustu_main"   0x00007ffff12257ed in __GI___select (nfds=23, readfds=0x7fffd18ac340 <mod_s>, writefds=0x7fff3fff6410, exceptfds=0x0, timeout=0x7fff3fff6400) at ../sysdeps/unix/sysv/linux/select.c:69
  72   Thread 0x7fff3effd640 (LWP 825285) "bambustu_main"   __futex_abstimed_wait_common64 (private=0, cancel=true, abstime=0x0, op=393, expected=0, futex_word=0x55555ed7ef00) at ./nptl/futex-internal.c:57
  671  Thread 0x7fff17e3d640 (LWP 825887) "bambustu_main"   0x00007ffff12257ed in __GI___select (nfds=57, readfds=0x7fff17e344f0, writefds=0x0, exceptfds=0x0, timeout=0x7fff17e344c0) at ../sysdeps/unix/sysv/linux/select.c:69
  703  Thread 0x7fff677fe640 (LWP 826396) "threaded-ml"     0x00007ffff1222d7f in __GI___poll (fds=0x7fff54004ef0, nfds=3, timeout=20) at ../sysdeps/unix/sysv/linux/poll.c:29
  704  Thread 0x7fff667fc640 (LWP 826397) "bambustu_main"   0x00007ffff11ef868 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=0x7fff667f3490, rem=0x7fff667f3480) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78
* 705  Thread 0x7fffe8ebc640 (LWP 826399) "source:src"      ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
(gdb) bt
#0  ____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
#1  0x128595952294ce44 in  ()
(gdb) record function-call-history
warning: Decode error (-23) at instruction 26926 (offset = 0xfc0, pc = 0x555559c3ee3d): expected tracing enabled event.
warning: Decode error (-6) at instruction 104520 (offset = 0x3fc7, pc = 0x7ffff7fea2ba): unexpected packet context.
6670	g_mutex_unlock
6671	??
6672	gst_pad_set_chain_function_full@plt
6673	??
6674	_dl_runtime_resolve_xsavec
6675	_dl_fixup
6676	_dl_lookup_symbol_x
6677	do_lookup_x
6678	check_match
6679	strcmp
6680	[decode error (-6): unexpected packet context]
(gdb) 
```

lanewei120, 2023-02-03:
@jwise 
I just removed the jpeg decoder of gstreamer in /lib/x86_64-linux-gnu/gstreamer-1.0/libgstjpeg.so

and found it will find /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstnvcodec.so instead
and this time it works well

so it seems the libgstjpeg.so on ubuntu doesn't work well with our jpeg stream

jwise, 2023-02-03:
Very interesting.  I think the message `[decode error (-6): unexpected packet context]` is a complaint from gdb about something internal to the processor trace format (probably you shouldn't need gstreamer debug libraries)... I wonder why that happens.

Maybe try `(gdb) record btrace bts` instead of `(gdb) record btrace`?  Branch trace recording might be more robust than Intel Processor Trace.

`libgstnvcodec` might work on your machine but probably not on machines that don't have nvidia GPUs.

lanewei120, 2023-02-07:
@jwise 

Thread 711 "source:src" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc7fff640 (LWP 960248)]
____longjmp_chk () at ../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S:167
167	../sysdeps/unix/sysv/linux/x86_64/____longjmp_chk.S: No such file or directory.
(gdb) record function-call-history
839	jpeg_CreateDecompress
840	??
841	??
842	??
843	__longjmp_chk@plt
844	__longjmp_chk
845	_longjmp_unwind
846	__GI___pthread_cleanup_upto
847	__longjmp_chk
848	____longjmp_chk

lanewei120, 2023-02-07:
it seems when decoding the jpeg data by using libjpeg
it found some error
and we need to pass an error_handler to setjmp
if not set the error handler
it will exit or crash?

lanewei120, 2023-02-07:
the jpeg decoder in my gstreamer:
Factory Details:
  Rank                     primary (256)
  Long-name                JPEG image decoder
  Klass                    Codec/Decoder/Image
  Description              Decode images from JPEG format
  Author                   Wim Taymans <wim@fluendo.com>

Plugin Details:
  Name                     jpeg
  Description              JPeg plugin library
  Filename                 /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstjpeg.so
  Version                  1.20.3
  License                  LGPL
  Source module            gst-plugins-good
  Source release date      2022-06-15
  Binary package           GStreamer Good Plugins (Ubuntu)
  Origin URL               https://launchpad.net/distros/ubuntu/+source/gst-plugins-good1.0

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstVideoDecoder
                         +----GstJpegDec

Pad Templates:
  SINK template: 'sink'
    Availability: Always
    Capabilities:
      image/jpeg
  
  SRC template: 'src'
    Availability: Always
    Capabilities:
      video/x-raw
                 format: { (string)I420, (string)RGB, (string)BGR, (string)RGBx, (string)xRGB, (string)BGRx, (string)xBGR, (string)GRAY8 }
                  width: [ 1, 2147483647 ]
                 height: [ 1, 2147483647 ]
              framerate: [ 0/1, 2147483647/1 ]

Element has no clocking capabilities.
Element has no URI handling capabilities.

Pads:
  SINK: 'sink'
    Pad Template: 'sink'
  SRC: 'src'
    Pad Template: 'src'

Element Properties:
  automatic-request-sync-point-flags: Flags to use when automatically requesting sync points
                        flags: readable, writable
                        Flags "GstVideoDecoderRequestSyncPointFlags" Default: 0x00000003, "corrupt-output+discard-input"
                           (0x00000001): discard-input    - GST_VIDEO_DECODER_REQUEST_SYNC_POINT_DISCARD_INPUT
                           (0x00000002): corrupt-output   - GST_VIDEO_DECODER_REQUEST_SYNC_POINT_CORRUPT_OUTPUT
  automatic-request-sync-points: Automatically request sync points when it would be useful
                        flags: readable, writable
                        Boolean. Default: false
  discard-corrupted-frames: Discard frames marked as corrupted instead of outputting them
                        flags: readable, writable
                        Boolean. Default: false
  idct-method         : The IDCT algorithm to use
                        flags: readable, writable
                        Enum "GstIDCTMethod" Default: 1, "ifast"
                           (0): islow            - Slow but accurate integer algorithm
                           (1): ifast            - Faster, less accurate integer method
                           (2): float            - Floating-point: accurate, fast on fast HW
  max-errors          : (Deprecated) Error out after receiving N consecutive decoding errors (-1 = never fail, 0 = automatic, 1 = fail on first error)
                        flags: readable, writable, deprecated
                        Integer. Range: -1 - 2147483647 Default: 0 
  min-force-key-unit-interval: Minimum interval between force-keyunit requests in nanoseconds
                        flags: readable, writable
                        Unsigned Integer64. Range: 0 - 18446744073709551615 Default: 0 
  name                : The name of the object
                        flags: readable, writable, 0x2000
                        String. Default: "jpegdec0"
  parent              : The parent of the object
                        flags: readable, writable, 0x2000
                        Object of type "GstObject"
  qos                 : Handle Quality-of-Service events from downstream
                        flags: readable, writable
                        Boolean. Default: true

jwise, 2023-02-07:
Ohhhhh, I bet I know what is going on.  I wonder if the libjpeg that Gstreamer is linking against is incompatible with the libjpeg built into Bambu Studio.  The libjpeg built into Bambu Studio is rather old; my system has libjpeg.so.8.2.2 (jconfig.h header says 2.1.2), but Bambu Studio links against, apparently, "JPEG/2.0.6.zip".

I wonder if we're mixing symbols or something, and the internals of the libjpegs are different.  Should we instead dynamically link against the system libjpeg?

lanewei120, 2023-02-07:
but the libjpeg in bambu_studio is not used in this cases
do you mean when gstreamer's libjpegdec.so links to libjpeg.so
it will link to the libjpeg static built in bambu_studio?

jwise, 2023-02-07:
I suspect that is what is happening, but I don't know for sure.  (You could probably find out with `record instruction-history`, if you wanted to go diving into a disassembly.)  The beginning of `jpeg_CreateDecompress` certainly suggests this as a possibility:

```
GLOBAL(void)
jpeg_CreateDecompress(j_decompress_ptr cinfo, int version, size_t structsize)
{
  int i;

  /* Guard against version mismatches between library and caller. */
  cinfo->mem = NULL;            /* so jpeg_destroy knows mem mgr not called */
  if (version != JPEG_LIB_VERSION)
    ERREXIT2(cinfo, JERR_BAD_LIB_VERSION, JPEG_LIB_VERSION, version);
```

In my build of `jdapimin.c`, it looks like here is the corresponding set of instructions:

```
0000000000000000 <jpeg_CreateDecompress>:
   0:   f3 0f 1e fa             endbr64 
   4:   55                      push   %rbp
   5:   48 89 d5                mov    %rdx,%rbp
   8:   53                      push   %rbx
   9:   48 89 fb                mov    %rdi,%rbx
   c:   48 83 ec 08             sub    $0x8,%rsp
  10:   48 c7 47 08 00 00 00    movq   $0x0,0x8(%rdi)
  17:   00 
  18:   83 fe 3e                cmp    $0x3e,%esi
  1b:   74 13                   je     30 <jpeg_CreateDecompress+0x30>
  1d:   48 8b 07                mov    (%rdi),%rax
  20:   48 8b 15 00 00 00 00    mov    0x0(%rip),%rdx        # 27 <jpeg_CreateDecompress+0x27>
  27:   89 70 30                mov    %esi,0x30(%rax)
  2a:   48 89 50 28             mov    %rdx,0x28(%rax)
  2e:   ff 10                   call   *(%rax)
```

So, I guess, if you hit the `call` that happens right after the `cmp $0x3e`, then that's probably a sign that you have a jpeg API version mismatch.

lanewei120, 2023-02-07:
I dumped a jpeg from gstbambusrc.c:gst_bambusrc_create
and I can play it using following command on the same pc:
gst-launch-1.0 multifilesrc location=1.jpeg loop=true ! jpegdec ! ximagesink
which will use the same libgstjpeg.so

so it seems you are right

I also tried to build the libjpeg.a in deps of bambu_studio
using version 2.15, but it seems no use

I will try to change the libjpeg to dynamic for a try

lanewei120, 2023-02-07:
oops, just missed one flag(WITH_JPEG8)  when building 2.1.5
and after add this flag, it can work now
--- a/deps/JPEG/JPEG.cmake
+++ b/deps/JPEG/JPEG.cmake
@@ -1,8 +1,11 @@
 bambustudio_add_cmake_project(JPEG
-    URL https://github.com/libjpeg-turbo/libjpeg-turbo/archive/refs/tags/2.0.6.zip
-    URL_HASH SHA256=017bdc33ff3a72e11301c0feb4657cb27719d7f97fa67a78ed506c594218bbf1
+    URL https://github.com/libjpeg-turbo/libjpeg-turbo/archive/refs/tags/2.1.5.zip
+    #URL https://github.com/libjpeg-turbo/libjpeg-turbo/archive/refs/tags/2.0.6.zip
+    URL_HASH SHA256=7118e702c8881880812d54e0232f20248707e1da418858323f68268a70ff3f8e
+    #URL_HASH SHA256=017bdc33ff3a72e11301c0feb4657cb27719d7f97fa67a78ed506c594218bbf1
     DEPENDS ${ZLIB_PKG}
     CMAKE_ARGS
         -DENABLE_SHARED=OFF
         -DENABLE_STATIC=ON
+        -DWITH_JPEG8=ON
 )

so I just checked my libjpeg.so on ubuntu
it is using libjpeg.so.8

maybe we need to add a detect logic of jpeg version before building the deps

jwise, 2023-02-07:
Oh, I see why that works:

```
if(WITH_JPEG8)
  set(JPEG_LIB_VERSION 80)
elseif(WITH_JPEG7)
  set(JPEG_LIB_VERSION 70)
else()
  set(JPEG_LIB_VERSION 62)
endif()
```

So I guess that is, in theory, ABI compatible.

I dunno.  It still makes me uneasy :) It seems like it solves the problem for this system, but could break on other systems in the other direction (i.e., if a system gstreamer links against jpeg6, then this breaks).  I think I continue to advocate for using a dynamically linked libjpeg, if possible.

SaltWei, 2023-03-01:
Included in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei