bambulab/BambuStudio #1190: To change chamber fan percentage i need to open and close the panel many times

pitops, 2023-01-27:
**Describe the bug**
For some reason my printer keeps running at 100% chamber fan for ALL prints EVERY time. (printing PLA btw)

I go to manually reduce it after it starts getting to 100%. However with the latest bambu studio version **1.4.2.13** this became even more cumbersome.

First of all it takes time for the percentage to show in the settings as seen below all are zero.

[image]

[image]

After the stats come back. When i click on **-** it closes the panel and have to reopen to further reduce the speed from 100% to 50%.

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to **Device** tab then see screenshots above

**Expected behavior**
1. Be able to change the chamber speed much faster (even be able to add my own value instead of requiring 5 clicks to go from 100% to 50%).
2. Avoid closing the panel when i hit **-** or **+**
3. If the chamber fan goes to 100% because of bambu studio and not because of printer firmware also fix that.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Version 12.6

walterwongbbl, 2023-02-01:
Hi.
This is a bug,     +or - operation should not close the window. 
 we will fix this issue.

rkarlsba, 2023-02-09:
I see the same here with BS 1.4.2.13 and X1 Carbon with 01.04.00.00 firmware. If I click a fan, the three fan controls show up and if I choose to change one of them, the dialog closes and that fan is changed. I cannot find any way to set the chamber fan in the job either, so this must be done in realtime, making it pretty far from convenient. Setting the wanted chamber temperature also doesn't work or is simply ignored. We need to be able to control these things, really. The defaults make nice headlines, but they don't make a winner!

walterwongbbl, 2023-02-10:
> I see the same here with BS 1.4.2.13 and X1 Carbon with 01.04.00.00 firmware. If I click a fan, the three fan controls show up and if I choose to change one of them, the dialog closes and that fan [...]

Hi.
We will fix this issue. Can you provide a video of the operation. After clicking the button(+ or -), the fan control window will be closed.

rkarlsba, 2023-02-10:
> open and the panel many times (Issue #1190)

There's no need to make a video. Just open BS and click Device, then click one of the fan icons on the right, below the speed switch and the lights switch. Then attempt to change the speed on one of the fans and it'll close the dialogue, usually, but not always, changing the speed after closing it. It is NO way to set a predefined chamber temperature, probably because there isn't a chamber heater, only the bed and hotend heaters. Also, there are fans that can be used to control the amount of air going through the printer at any time, thus making it quite easy to control the chamber temperature. 

If this printer was open source, these bugs would have been fixed ages ago, but since you have chosen to stay closed, we customers must all beg you to fix them instead of just fixing the bugs ourselves and offering a pull request. The mentioned method works very well in thousands of open source systems and would work just as well here, leaving your precious workers' time to be used for other things. 

So please, can you open up now once and for all so that we don't get all this time waiting for someone to fix trivial bugs? 

Vennlig hilsen 

roy 
-- 
Roy Sigurd Karlsbakk 
(+47) 98013356 
-- 
I all pedagogikk er det essensielt at pensum presenteres intelligibelt. Det er et elementært imperativ for alle pedagoger å unngå eksessiv anvendelse av idiomer med xenotyp etymologi. I de fleste tilfeller eksisterer adekvate og relevante synonymer på norsk.

DapperDangus, 2023-02-10:
You bought a closed source knowing so maybe don’t complain to the company working their *** off to deliver everything you guys want. Dear lord people, don’t buy a Mac then expect to modify it like a PC.

rkarlsba, 2023-02-10:
I just suggest we could help out fixing the bugs they aren't able to fix if they were to open their systems. 

PS: A PC is a common word for any sort of computer. A Mac is a peculiar thing that can run mostly all sorts of OSes, but also macOS. 

Vennlig hilsen 

roy 
-- 
Roy Sigurd Karlsbakk 
(+47) 98013356 
-- 
I all pedagogikk er det essensielt at pensum presenteres intelligibelt. Det er et elementært imperativ for alle pedagoger å unngå eksessiv anvendelse av idiomer med xenotyp etymologi. I de fleste tilfeller eksisterer adekvate og relevante synonymer på norsk.

DapperDangus, 2023-02-10:
In America it’s PC vs Mac but thanks…

Keep up the good work Bambu.

walterwongbbl, 2023-02-11:
Hi.
Thank you for your suggestion. We have fixed this issue and will add it to the next version

cameo69, 2023-02-13:
Please fix this. Also the fact that you have to wait until the current value has been loaded again.
[image]
BambuStudio v1.4.2.13 on MacOS 13.1

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei

magnusmyklebost, 2023-04-04:
> Available in v1.5.0 Public Beta

When will this be added to the non beta? Im only using PLA and reducing chamber fan speed from 100% to 70% reduses the noise drastically. Ive never had any print issues.

I would like to be able to add a chamber fan value to the filament properties. Best would be to add it here, where you could set chamber fan, and also a field to "Automatic adjust fan speed if temperature goes above XXX degree" :
[image]

Felix-Quehl, 2023-05-06:
# Control Fan Speed in gcode of nozzle settings

[image]

## M106 command

You can overwrite the camber fan speed with the gcode of the nozzle settings inside Bambu Studio:

* The `M106` command sets a fan speed
* The `P` parameter is used to specify the fan to control by the assigned index of firmware
* The last parameter is fan speed S0 to S255 which equals 0% to 100%

### Syntax

```gcode
M106 PX SXXX ; set fan PX (0-n) to speed SXXX (0-255)
```

### Samples

```gcode
M106 P3 S0 ; set chamber fan to 0% 
M106 P3 S128 ; set chamber fan to 50%
M106 P3 S255 ; set chamber fan to 100%
``` 

[image]

JoshCu, 2023-05-13:
I found that I also needed to change the filament specific g-code profile:
[image]

Oliver-S3, 2023-06-18:
Just want to add to this. Printing ASA multi color. Tweaked the fan speed to keep the temp where I want it without the smell. Every time it switches filament it resets back to 0 based on the filament gcode. some sort of override would be nice so that you can adjust the chamber fan without having to reslice and cancel the print.

michelem09, 2023-06-22:
I have a similar issue with BS 1.6.4.2, when I try to set chamber fan, the value come back to the previous one. If you click one time on minus at 100% you got 90 for few seconds then it comes back to 100 and nothing happens fan side. Moreover it seems only few values work on the fan, 30/60/100 or something like that, adjusting it by 10% is impossible.

jomiller97, 2023-09-03:
How would I set if I wanted to change more than just the chamber fan speed? When I try to make 3 lines just changing the P3 to P1 and P2, it will change the last entry, but then set the other 2 to 0

lgarron, 2024-10-16:
I'm also finding that this is a big annoyance with my new Bambu printer. It prints excellently and quietly with 70% chamber fan speed, but changing the filament resets the chamber fan to a very noisy 100%.

> Available in v1.5.0 Public Beta

I'm still catching up on this issue, but I'm not quite sure what to make of this. I'm on 1.9.7.52 stable, does this mean this feature should be available? I'm not finding it anywhere in the UI.

`during_print_exhaust_fan_speed` in `fdm_filament_common.json` is set to `70` and the value is not overridden in `fdm_filament_pla.json` and yet I'm still getting 100% fan speed. I can see [some relevant code](https://github.com/bambulab/BambuStudio/blob/4d64b2d65249c141eac0e3cb43712989622012c7/src/libslic3r/GCode.cpp#L1984-L2004) but not any UI for it. Is the only option to modify the G-code as others have suggested above?