bambulab/BambuStudio #1196: Linux: crash sighting after submitting job to printer

jwise, 2023-01-28:
I saw the following crash right after 'changing to device screen in 3 seconds...'.  Unfortunately I did not compile wxWidgets with address sanitizer or with debug symbols, but at least I'm going to leave it here in case we reproduce it later.

```
==3099597==ERROR: AddressSanitizer: SEGV on unknown address 0x000079004dbb (pc 0x561e2038ae2c bp 0x615000ed9700 sp 0x7fffa2dfe590 T0)
==3099597==The signal is caused by a READ memory access.
    #0 0x561e2038ae2c in wxWebRequestImpl::ProcessStateEvent(wxWebRequest::State, wxString const&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa509e2c)
    #1 0x561e207ae1ef in wxEvtHandler::TryHereOnly(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92d1ef)
    #2 0x561e207ae23e in wxEvtHandler::ProcessEventLocally(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92d23e)
    #3 0x561e207ae2e8 in wxEvtHandler::ProcessEvent(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92d2e8)
    #4 0x561e207adaf2 in wxEvtHandler::ProcessPendingEvents() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92caf2)
    #5 0x561e206c32c6 in wxAppConsoleBase::ProcessPendingEvents() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa8422c6)
    #6 0x561e205bc5d4 in wxApp::DoIdle() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa73b5d4)
    #7 0x561e205bc6c6 in wxapp_idle_callback (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa73b6c6)
    #8 0x7f60d837cc43 in g_main_dispatch ../../../glib/gmain.c:3419
    #9 0x7f60d837cc43 in g_main_context_dispatch ../../../glib/gmain.c:4137
    #10 0x7f60d83d16c7 in g_main_context_iterate.constprop.0 ../../../glib/gmain.c:4213
    #11 0x7f60d837c2b2 in g_main_loop_run ../../../glib/gmain.c:4413
    #12 0x7f60d8cc2cfc in gtk_main ../../../../gtk/gtkmain.c:1329
    #13 0x561e205cd404 in wxGUIEventLoop::DoRun() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa74c404)
    #14 0x561e206f6e30 in wxEventLoopBase::Run() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa875e30)
    #15 0x561e20624950 in wxDialog::ShowModal() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa7a3950)
    #16 0x561e1a5c83d2 in Slic3r::GUI::DPIAware<wxDialog>::ShowModal() /home/joshua/work/BambuStudio/src/slic3r/GUI/GUI_Utils.hpp:232
    #17 0x561e1ae4c408 in Slic3r::GUI::Plater::priv::on_action_print_plate(Slic3r::GUI::SimpleEvent&) /home/joshua/work/BambuStudio/src/slic3r/GUI/Plater.cpp:6034
    #18 0x561e207adc55 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92cc55)
    #19 0x561e207ae03d in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92d03d)
    #20 0x561e207ae193 in wxEvtHandler::TryHereOnly(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92d193)
    #21 0x561e207ae23e in wxEvtHandler::ProcessEventLocally(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92d23e)
    #22 0x561e207ae2e8 in wxEvtHandler::ProcessEvent(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92d2e8)
    #23 0x561e207adaf2 in wxEvtHandler::ProcessPendingEvents() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa92caf2)
    #24 0x561e206c32c6 in wxAppConsoleBase::ProcessPendingEvents() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa8422c6)
    #25 0x561e205bc5d4 in wxApp::DoIdle() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa73b5d4)
    #26 0x561e205bc6c6 in wxapp_idle_callback (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa73b6c6)
    #27 0x7f60d837cc43 in g_main_dispatch ../../../glib/gmain.c:3419
    #28 0x7f60d837cc43 in g_main_context_dispatch ../../../glib/gmain.c:4137
    #29 0x7f60d83d16c7 in g_main_context_iterate.constprop.0 ../../../glib/gmain.c:4213
    #30 0x7f60d837c2b2 in g_main_loop_run ../../../glib/gmain.c:4413
    #31 0x7f60d8cc2cfc in gtk_main ../../../../gtk/gtkmain.c:1329
    #32 0x561e205cd404 in wxGUIEventLoop::DoRun() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa74c404)
    #33 0x561e206f6e30 in wxEventLoopBase::Run() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa875e30)
    #34 0x561e206c3c49 in wxAppConsoleBase::OnRun() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa842c49)
    #35 0x561e20749d56 in wxEntry(int&, wchar_t**) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa8c8d56)
    #36 0x561e1aaebb2d in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) /home/joshua/work/BambuStudio/src/slic3r/GUI/GUI_Init.cpp:64
    #37 0x561e16d7ad09 in Slic3r::CLI::run(int, char**) /home/joshua/work/BambuStudio/src/BambuStudio.cpp:452
    #38 0x561e16c49528 in main /home/joshua/work/BambuStudio/src/BambuStudio.cpp:2624
    #39 0x7f60d1a48d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #40 0x7f60d1a48e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #41 0x561e16d49064 in _start (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xec8064)
```

jwise, 2023-01-30:
I can reliably reproduce this by sending two print jobs in a row.  I rebuilt wxWidgets with address sanitizer, and it gave me:

```
==3398248==ERROR: AddressSanitizer: heap-use-after-free on address 0x615000c96600 at pc 0x563ed4748537 bp 0x7ffcc09db9b0 sp 0x7ffcc09db9a0
READ of size 8 at 0x615000c96600 thread T0
    #0 0x563ed4748536 in wxWebRequestImpl::ProcessStateEvent(wxWebRequest::State, wxString const&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa754536)
    #1 0x563ed54eb9f9 in wxEvtHandler::TryHereOnly(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f79f9)
    #2 0x563ed54eba92 in wxEvtHandler::ProcessEventLocally(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f7a92)
    #3 0x563ed54ebbda in wxEvtHandler::ProcessEvent(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f7bda)
    #4 0x563ed54ea341 in wxEvtHandler::ProcessPendingEvents() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f6341)
    #5 0x563ed51bfe68 in wxAppConsoleBase::ProcessPendingEvents() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb1cbe68)
    #6 0x563ed4e82bb0 in wxApp::DoIdle() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xae8ebb0)
    #7 0x563ed4e82ec6 in wxapp_idle_callback (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xae8eec6)
    #8 0x7fc650b07c43 in g_main_dispatch ../../../glib/gmain.c:3419
    #9 0x7fc650b07c43 in g_main_context_dispatch ../../../glib/gmain.c:4137
    #10 0x7fc650b5c6c7 in g_main_context_iterate.constprop.0 ../../../glib/gmain.c:4213
    #11 0x7fc650b072b2 in g_main_loop_run ../../../glib/gmain.c:4413
    #12 0x7fc65144dcfc in gtk_main ../../../../gtk/gtkmain.c:1329
    #13 0x563ed4eb0dfc in wxGUIEventLoop::DoRun() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xaebcdfc)
    #14 0x563ed527f58b in wxEventLoopBase::Run() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb28b58b)
    #15 0x563ed4fcede0 in wxDialog::ShowModal() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xafdade0)
    #16 0x563ece8b06d2 in Slic3r::GUI::DPIAware<wxDialog>::ShowModal() /home/joshua/work/BambuStudio/src/slic3r/GUI/GUI_Utils.hpp:232
    #17 0x563ecf134708 in Slic3r::GUI::Plater::priv::on_action_print_plate(Slic3r::GUI::SimpleEvent&) /home/joshua/work/BambuStudio/src/slic3r/GUI/Plater.cpp:6034
    #18 0x563ed54ea704 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f6704)
    #19 0x563ed54eb5c2 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f75c2)
    #20 0x563ed54eb8f2 in wxEvtHandler::TryHereOnly(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f78f2)
    #21 0x563ed54eba92 in wxEvtHandler::ProcessEventLocally(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f7a92)
    #22 0x563ed54ebbda in wxEvtHandler::ProcessEvent(wxEvent&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f7bda)
    #23 0x563ed54ea341 in wxEvtHandler::ProcessPendingEvents() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb4f6341)
    #24 0x563ed51bfe68 in wxAppConsoleBase::ProcessPendingEvents() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb1cbe68)
    #25 0x563ed4e82bb0 in wxApp::DoIdle() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xae8ebb0)
    #26 0x563ed4e82ec6 in wxapp_idle_callback (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xae8eec6)
    #27 0x7fc650b07c43 in g_main_dispatch ../../../glib/gmain.c:3419
    #28 0x7fc650b07c43 in g_main_context_dispatch ../../../glib/gmain.c:4137
    #29 0x7fc650b5c6c7 in g_main_context_iterate.constprop.0 ../../../glib/gmain.c:4213
    #30 0x7fc650b072b2 in g_main_loop_run ../../../glib/gmain.c:4413
    #31 0x7fc65144dcfc in gtk_main ../../../../gtk/gtkmain.c:1329
    #32 0x563ed4eb0dfc in wxGUIEventLoop::DoRun() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xaebcdfc)
    #33 0x563ed527f58b in wxEventLoopBase::Run() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb28b58b)
    #34 0x563ed51c2d80 in wxAppConsoleBase::OnRun() (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb1ced80)
    #35 0x563ed539bdc5 in wxEntry(int&, wchar_t**) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xb3a7dc5)
    #36 0x563ecedd3e2d in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) /home/joshua/work/BambuStudio/src/slic3r/GUI/GUI_Init.cpp:64
    #37 0x563ecb063009 in Slic3r::CLI::run(int, char**) /home/joshua/work/BambuStudio/src/BambuStudio.cpp:452
    #38 0x563ecaee36c8 in main /home/joshua/work/BambuStudio/src/BambuStudio.cpp:2624
    #39 0x7fc64a1d3d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #40 0x7fc64a1d3e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #41 0x563ecb031364 in _start (/home/joshua/work/BambuStudio/build/src/bambu-studio+0x103d364)

0x615000c96600 is located 0 bytes inside of 480-byte region [0x615000c96600,0x615000c967e0)
freed by thread T0 here:
    #0 0x7fc65205d22f in operator delete(void*, unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172
    #1 0x563ed47440e1 in wxWebRequest::operator=(wxWebRequest const&) (/home/joshua/work/BambuStudio/build/src/bambu-studio+0xa7500e1)

previously allocated by thread T0 here:
    #0 0x7fc65205c1c7 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:99
[... cut ...]
I think that the correct way to do this is that you must not store a `wxWebSession`, but you must instead always store a `wxWebSession *`, and then use `wxWebSession::IncRef` and `wxWebSession::DecRef` to track your own usage of it.  (I think this is wrong on every platform, not just on Linux, but somehow it doesn't seem to result in a crash on other platforms.)

It is bed time for me so I don't have time to write a fix for it right now, but this should be enough to pinpoint the problem.  This pattern also appears in `BindDialog` and `StatusPanel`, so you may wish to fix those there, as well.

walterwongbbl, 2023-01-30:
Hi. thank you very much everything you do.  We will fix this issue as soon as possible.

jasoncouture, 2023-02-06:
I have the same issue: https://github.com/bambulab/BambuStudio/issues/760#issuecomment-1418444434

Possible duplicate: #798

c2h5oh, 2023-02-06:
Can confirm - same problem using bambustudio 01.04.02.13-1 aur on arch.

jwise, 2023-02-07:
OK so there's good news and there's bad news.  The good news is that, apparently, even better than calling `IncRef` and `DecRef` by hand, there's such thing as a `wxObjectDataPtr`.

The bad news is that a `wxWebRequest` is, itself, not a `wxRefCounter`.  This might actually be a wxWidgets bug, where the `impl` doesn't stick around long enough to live into the `CallAfter`?  I thought this was going to be an easy solution for tonight, but it looks like this one gets pushed off once again for another night :)

lanewei120, 2023-02-08:
we just submitted a fix for the crash when switch between printer(some also happens after print/send_to_print)
the issue with above callstack is very likely fixed by this patch
https://github.com/bambulab/BambuStudio/commit/9fddc49787e7af46b2af2538fe8746fc0fb6e14d

you can also have a try

currently we found another crash when sending to printer, the stack is like this:
```
Thread 730 "bambustu_main" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff080df640 (LWP 1065710)]
0x00007fffe81cdb55 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() () from /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
(gdb) bt
#0  0x00007fffe81cdb55 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#1  0x00007fffe81af817 in BBL::AccountManager::start_send_gcode_to_sdcard(BBL::PrintParams, std::function<void (int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>, std::function<bool ()>)::{lambda(BBL::Sftp::Progress, bool&)#4}::operator()(BBL::Sftp::Progress, bool&) const [clone .isra.0] [clone .cold] ()
    at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#2  0x00007fffe81f7f28 in BBL::Sftp::priv::xfercb(void*, long, long, long, long) () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#3  0x00007fffe82f4c7b in Curl_pgrsUpdate () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#4  0x00007fffe82f22d2 in multi_runsingle () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#5  0x00007fffe82f3737 in curl_multi_perform () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#6  0x00007fffe82e44eb in curl_easy_perform () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#7  0x00007fffe81f94fa in BBL::Sftp::priv::sftp_perform() () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#8  0x00007fffe8239a93 in BBL::AccountManager::start_send_gcode_to_sdcard(BBL::PrintParams, std::function<void (int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>, std::function<bool ()>) () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#9  0x00007fffe829638a in BBL::BambuNetworkAgent::start_send_gcode_to_sdcard(BBL::PrintParams, std::function<void (int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>, std::function<bool ()>) () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#10 0x00007fffe829ec34 in bambu_network_start_send_gcode_to_sdcard () at /home/lanewei/.config/BambuStudio/plugins/[libbambu_networking.so](http://libbambu_networking.so/)
#11 0x00005555573951fc in Slic3r::NetworkAgent::start_send_gcode_to_sdcard(BBL::PrintParams, std::function<void (int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>, std::function<bool ()>) (this=0x55555cbc7aa0, params=..., update_fn=..., cancel_fn=...)
    at /home/lanewei/codes/github_branch/bamboo_slicer/src/slic3r/Utils/NetworkAgent.cpp:863
#12 0x00005555574e299e in Slic3r::GUI::SendJob::process() (this=0x555561687ad0) at /home/lanewei/codes/github_branch/bamboo_slicer/src/slic3r/GUI/Jobs/SendJob.cpp:276
#13 0x0000555557158f15 in Slic3r::GUI::Job::run(std::__exception_ptr::exception_ptr&) (this=0x555561687ad0, eptr=...) at /home/lanewei/codes/github_branch/bamboo_slicer/src/slic3r/GUI/Jobs/Job.cpp:14
#14 0x000055555715972c in operator()() const (__closure=0x55555e520768) at /home/lanewei/codes/github_branch/bamboo_slicer/src/slic3r/GUI/Jobs/Job.cpp:113
#15 0x000055555715b4e4 in boost::detail::thread_data<Slic3r::GUI::Job::start()::<lambda()> >::run(void) (this=0x55555e520630) at /home/lanewei/codes/dep_linux_ubuntu/usr/local/include/boost-1_75/boost/thread/detail/thread.hpp:120
#16 0x00005555579c581b in thread_proxy ()
#17 0x00007ffff118bb43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#18 0x00007ffff121da00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

we will fix this issue soon

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei