bambulab/BambuStudio #1199: Allow change of filament color printing order, especially on the first layer.

randyre, 2023-01-28:
Allow change of filament color printing order, especially on the first layer.

**Describe the solution you'd like**
Allow the user to define which color prints first. 

**Describe alternatives you've considered**
None.

I currently have a two-color print and the lighter color prints first, but as the second color prints, the nozzle barely touches the light color and it then streaks onto the darker color. Especially silk colors are very prone to this.

[image]

randyre, 2023-02-21:
Any way to fool the software to print one object before the other? this project is on hold until I can print the darker color first.

javawizard, 2023-05-07:
This would be a fantastic addition. I'm printing some parts that are 98% water soluble support on the first layer and 2% actual part and the actual part filament prints first, and the small features in said part don't turn out nearly as well as I imagine they would if they were surrounded by support filament when they were printed.

mike-serinus, 2023-07-31:
I second this issue. It is a major problem, particularly with narrow objects such as text on the first layer. If the text is not printed first, the surrounding color tends to spread into the text zone, making the text thinner than it should be. Compensating with line width does not help

darrenph1, 2023-08-02:
+1 We should be able to choose the color order of the layer. The definitely effects the outcome of the layer.

zhimin-zeng-bambulab, 2023-08-18:
It has been added in  1.7.4(https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52), you can have a try.

closed 2023-08-18 by zhimin-zeng-bambulab