bambulab/BambuStudio #1200: [Feature Request] Use the chamber camera to prevent head crashes

BambuLabDiscord, 2023-01-28:
nullIf the printer is equipped with the chamber camera, the printer should use the camera to prevent homing or other movements/processes that would possibly result in the print head crashing into the printed parts on the build plate.

Benderrodrigu3z, 2023-01-28:
you do know that you can turn on avoid printed parts in the slicer and then say avoid by how much right? @BambuLabDiscord

SaltWei, 2023-01-29:
Thank you very much for suggestions!
But this seams to be handled by printer and maybe official support can't be got timely here(For slicer).

I have transfered this ticket to printer team.

For future issue about firmware and printer, please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better support. Thanks!