bambulab/BambuStudio #1201: White Screen - home screen and dialog boxes

gtwaites, 2023-01-28:
BambuStudio was running perfectly until Windows 11 upgrade. Now home screen is blank (see attached) and dialog boxes are blank.

Configuration:
Intel Core i7-9700
32GB RAM
Windows 11 Pro, Build 22000.1516
Nvidia GeForce RTX 2060, Driver: 528.24
BambuStudio v01.04.02.13

[image]
[image]

SaltWei, 2023-01-29:
@gtwaites 
Could you zip all the log files in C:\Users\user.name\AppData\Roaming\BambuStudio\log and upload here?

we will have a check. Thank you very much!

StoneLiBambulab, 2023-02-15:
Fixed and the next version is coming soon.

Thanks!
Stone Li

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei

Drilmo, 2023-05-01:
Hi, 
I'm on last version and i have same error, i can't login...
[image]

chapterv, 2024-04-09:
[image]
not fixed

Helton-Loureiro, 2024-06-19:
I have the same problem
[image]