bambulab/BambuStudio #1207: Support Filament is being "Flush into objects' infill"

DasKashi, 2023-01-30:
**Describe the bug**
When using "Flush into objects' infill" with support filament, it also get's flushed into the object's infill which shouldn't be happening.

**3mf File for This Bug**
(https://www.printables.com/model/384420)

**To Reproduce**
Steps to reproduce the behavior:
1. use a model with supports
2. use support filament as support interface
3. activate "Flush into objects' infill"
4. Slice

**Expected behavior**
Either flatout exclude support filaments from being flushed into infill or give the option to manually exclude filament slots.

**Screenshots**
[image]

**Desktop:**
 - OS: Windows
 - Version 10

SaltWei, 2023-01-30:
@zhimin-zeng-bambulab Maybe we need to add special logic and never flush support filament into object.

zhimin-zeng-bambulab, 2023-01-31:
We will optimize the flushing logic so that the support material is not flushed to the fill or flushed to the object.

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei

austinwright, 2025-01-14:
@SaltWei - How exactly can the end user leverage this feature? I tried tracking back the `print_config.filament_is_support` to its usages to try to understand where that value is coming from. I then jumped to the BBL filament profiles, and only `fdm_filament_pva.json` has `filament_is_support` set to `true`. From there I noticed that the code for the filament basic settings tab, and the `s_Preset_filament_options` vector are using the same keys (`filament_soluble`/`filament_is_support`), but those options are not actually available in the filament options tab, even if advanced is toggled `on`.

Anyways, knowing Bambu PVA preset has the value set to `true`, I tested this functionality. I set the filament I want to use as support to Bambu PVA. What I noticed is that when I check the `flush into objects' infill` option, everything works as expected -> The support filament is not flushed into the sacrificial object at all. However, when I select the `flush into this object` option, the support filament **_is_** still flushed into the sacrificial object. This seems like a bug and/or oversight as part of fixing this issue.

UPDATE: While writing this up I decided to poke around some more and found 'develop mode' in the top-level Bambu Studio settings. Turning that on reveals the previously-hidden filament settings mentioned above. So I am able to test this again but with the support filament I actually have (printing PLA and using PETG as the support interface). Setting the filament to soluble and/or support in this way makes the filament behave in the same way as the Bambu PVA described above -> `flush into objects' infill` correctly omits the support filament, whereas `flush into this object` does not omit the support filament.