bambulab/BambuStudio #1208: [Issue] Bambu Studio not loading home page

BambuLabDiscord, 2023-01-30:
nullI am not sure when it starts, but at some point, when starting Bambu Studio, it takes up to 3 minutes to start up and the home page is completely blank. Once it starts, it seems to stay for a while until I reinstall.

All user presets are there and other cloud functionality works.

SaltWei, 2023-01-30:
A known issue and we ara solving.
Thank you very much!

StoneLiBambulab, 2023-04-24:
Does the new version still have this problem? It has been optimized.

Thanks!

Stone Li

Gingerbeardicus, 2023-08-22:
I have been going through this issue  during the last few releases. Currently on 1.7.3.50 running Windows 11 Pro 22H2. The loading takes significantly longer which correlates to the home page being blank. In order for me to temporarily resolve, I need to uninstall any recent software installs that seem completely unrelated. I am almost at the point of needing to run this in a sandbox or a vm just to save my frustration. 

Hope there is some progress on this soon. Such a great machine but the slicer issue really detracts from the overall experience. Please let me know if there is anything I can provide to support this resolution.

[image]

Xaeco, 2024-01-18:
I have the same issue. I'm running 
Edition: Windows 11 Pro
Version: 23H2
AMD Ryzen 9 7900 12-Core Processor

If I run Bambu Studio in Windows Sandbox it opens straight away but performance is not good enough and sandbox crashes often.

cardboardpig, 2024-10-24:
I have this issue as well. 

I can't add my X1C to the slicer because I cannot log in...

theboyknowsclass, 2025-01-04:
https://github.com/bambulab/BambuStudio/issues/3807 seems to fix it for me