bambulab/BambuStudio #1209: Excessive Input Shaper Smoothing due to relying on static slicer based acceleration values

ChaosBlades, 2023-01-30:
I have noticed the X1C has less detail than my Voron printers. I think this is because to uses slicer based static acceleration values that depending on the results of XY compensation could result in too much smoothing. On my Voron printers I run automatic Input Shaper Calibration then set max accelerations in firmware based on what it recommends. I then disable all acceleration settings in the slicer. With the X1C and Bambu Studio the slicer determines the Acceleration by default. I propose that by default the printer firmware determines the proper acceleration values based on the XY calibration and Bambu Studio has a toggle to override these firmware based values. When relying on static acceleration values prints can loose a lot of detail depending on the results of the XY calibration. The tooltip for this override could mention something about 'firmware based acceleration is more detailed but potentially slower vs. slicer based acceleration is faster but some loss of fine detail'.

Or am I mistaken and the X1C will set an acceleration hard cap based on XY compensation to prevent excessive smoothing that overrides the slicer settings?

SaltWei, 2023-01-30:
> I have noticed the X1C has less detail than my Voron printers.

Thank you for reporting issue to us!
Could you share some pictures and 3MF for debugging?

SaltWei, 2023-01-30:
response from printer team:
The less detail is from the accurate dynamic extrusion compensation (yes , because of the "accurate" ).
you can add "M900 L500.0 ; Empirical parameters" at the last line of the start gcode ( this will proximatly cut the compensation to about half ) and try again :
[image]

but this will have negative effect on the auto-slowdown region .
the root cause is the polymer flow dynamics when the melted polymer extruded out from the nozzle, which we are still investigating and trying to compensate.