bambulab/BambuStudio #1211: Linux: bambu_source shm_open lifecycle is kind of unpleasant on Linux

jwise, 2023-01-30:
The lifecycle for bambu_source has a handful of problems on Linux:
* For one, if `bambu_source` dies for any reason, then `/dev/shm/bambu_stream_url` remains until something goes and cleans it up (which nothing will ever do!) -- and then `bambu_source` will never start up again (it reports `[BambuTunnelCamera::open] shm_open failed 17`).
* For another, when it's time to stop streaming, `bambu_source` continues to send data on to ffmpeg, which then continue encoding in the background -- nothing ever will shut `bambu_source` down, and so `ffmpeg` will sit there burning CPU indefinitely.  (I think this is a cross-platform problem.)

I think the solution for this is probably:
* The `libBambuSource` should write its pid into the shm buffer (maybe the first 8 bytes?).  That way, it would be possible to look at the shm buffer and see if there really is a living `bambu_source` or not -- and if not, then it's safe to clean up the shm buffer.
* The `libBambuSource` should return `Bambu_stream_end` five minutes or so after the `url.txt` changes to being empty, which will allow ffmpeg to shut down gracefully.

(See #1210.)

bambu123, 2023-02-03:
@jwise Thanks for the report. I just fixed it yesterday.
Our bambu_source for Linux has not been published yet. It will come with the next Studio version 1.5.
But the problem is in libBambuSource.so in the network plugin, I would put a new one here and you can update it manually.

jwise, 2023-02-03:
@bambu123 Sounds good -- thanks!  If you want a `bambu_source` for Linux, I wrote one in #1210 that works for me :-)

bambu123, 2023-02-03:
@jwise Please try this

[libBambuSource.zip](https://github.com/bambulab/BambuStudio/files/10575309/libBambuSource.zip)

jwise, 2023-02-03:
Oop, I'd meant to have some time to test the new binaries tonight, but that didn't happen.  I will try it over the weekend and report back.  Thanks!

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei