bambulab/BambuStudio #1212: First Layer Inspection has been removed?

candreacchio, 2023-01-30:
**Describe the bug**
It seems that there is no longer any first layer inspection

**To Reproduce**
Steps to reproduce the behavior:
1. Open Linux Build of Bambu Studio
2. Send file to printer with all checkboxes ticked
3. double check on printer that first layer inspection is turned on
4. notice at the end of the first layer inspection doesnt occur

**Expected behavior**
That first layer inspection occurs.

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.10
 - Version: 60b665c4b57c4e9a63271a4072f239674cfa8e08 v1.4.2

tangketan, 2023-02-03:
Did you use textured PEI plate? First layer inspection is not disabled for textured PEI plate.

candreacchio, 2023-02-04:
I am using the textured PEI plate, any reason why its not supported?

tangketan, 2023-02-06:
The surface of many Textured PEI plates are not smooth enough for accurate first layer inspection. We have received several false alarms, so we decided to turn it off before we find the solution, which would be pushed from two sides:
- manufacturing side: improve the smoothness of Textured PEI surface;
- algorithm side: estimate the smoothness of build plate, only turn off first layer inspection when the smoothness is too low.

candreacchio, 2023-02-06:
Does that mean the Textured PEI surface that i have at the moment, will never be compatible with the first layer inspection?

I am also surprised as it still does flow calibration totally fine.

theogott, 2023-02-09:
If the LIDAR could scan the PEI Plate "as is" and then in a second step compare with the result after doing the test-print, a inspection should theoretically be possible, it may however be an AI Task :-). As you have to remove the "Noise from the plate".

tangketan, 2023-02-12:
@candreacchio For now, yes. Flow calibration is easier, requires less data and more robust against depth noise.
@theogott That's what we are planing to do :D

tangketan, 2023-08-02:
Flow dynamics calibration and first layer inspection on PEI plate are back with the latest firmware 1.6 and Bambu Studio 1.7. This issue is closed.

closed 2023-08-02 by tangketan