bambulab/BambuStudio #1217: Add a gcode section executed for each layers

DzzD, 2023-01-30:
For now, we can add gcode manually for a layer after slicing, it would be great and very usefull to be able to set a gcode that would be added/executed for each layer. 

Could help to make calibration towers : temps/PA,speed as well as many other usefull stuffs.

By using conditional operators based on layer height or index, it would then be possible to make very usefull little programs.

DzzD, 2023-03-09:
ho ! great is that new or do I missed this ?

EukaDesigns, 2023-03-09:
I'm not sure if it was always available or not. 

Where would you prefer it to be? I thought this worked out nicely since you can save multiple sets of printer settings in case you wanted to save multiple g-code settings.

DzzD, 2023-03-10:
I would prefer to be able to define it by part or plate, but it seems that the very latest release of Bambustudio update enable to make it per plate (at least it is on the release notes, did not tried it yet)

SaltWei, 2023-03-10:
> at least it is on the release notes

No. The release note says supporting different custom gcode inserted in preview page for different plate, rather than machine preset

DzzD, 2023-03-10:
[image]

FlowForm, 2023-03-21:
Yes I'm also missing the "SpeedTower" scripts from Cura....

Could anyone make a script/code for changing speed at different layers, so I could insert it in the "machine gcode" section? (10mm/s increments for each 10 layer would be very useful)

closed 2023-04-01 by DzzD