bambulab/BambuStudio #1219: zoom with the many to the cam on bambu handy

Looker1981, 2023-01-31:
in my opinion if you make sure that you can zoom with the many to the cam on bambu handy it's a spectacular thing

SaltWei, 2023-02-08:
This is not slicer issue.

For bambu Handy issue, please move to https://github.com/bambulab/Bambu-Handy

closed 2023-02-08 by SaltWei

Looker1981, 2023-02-08:
I understand but in any case I ask this to ensure that it is updated if you can send this communication to improve

SaltWei, 2023-02-08:
Have transferred this ticket to APP team internally

Looker1981, 2023-02-08:
thanks anyway I hope you make the Italian version we are a group of about 500 members of an Italian channel that we have the bambulab please make a language also in Italian of the slicer and printer

SaltWei, 2023-02-08:
Italian is in our plan（studio） now.

Looker1981, 2023-02-08:
I don’t see Italian language … where can I find it Italian?

SaltWei, 2023-02-08:
Just plan of studio

Looker1981, 2023-02-08:
when did you update?

SaltWei, 2023-02-08:
No exact time. We are finding translator.
Will update as soon as possible when ready

Looker1981, 2023-02-08:
please tell him to do it in italian also for the printer and bambu handy