bambulab/BambuStudio #1222: Different color first-layer text fails

holla2040, 2023-01-31:
Added a single layer text in a different color to the base layer. Text prints first slowly, filament changed to print remaining first layer. Second color doesn't print slowing, even though its one the first later. Second color doesn't adhere to bed.

**3mf File for This Bug**
Attached

**To Reproduce**
Open attached 3MF file. Print in 2 colors.

**Expected behavior**
When the second color starts printing, it never adheres to the plate.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version 19045.2486
- Bambu Studio 1.4.2.13

[image]
[image]

[snifferTop.zip](https://github.com/bambulab/BambuStudio/files/10543601/snifferTop.zip)

morgano86, 2023-01-31:
I just loaded your file and I note your numbers aren't flush with the first layer. They are projecting outwards - at least according to the slicer. Therefore the first layer printed is the numbers and the 2nd layer is on top of the numbers, therefore, floating in the air. I think you need to adjust the model so the numbers are flush.

holla2040, 2023-01-31:
That makes sense, I tried to get them flush. Apparently, I need to work on that. I'll start first thing this morning. If I'm successful, I'll close this issue.
Thanks for your fast reply.

holla2040, 2023-01-31:
@morgano86 You were correct, I got it working after tweaking the z positions. Thanks.

holla2040, 2023-01-31:
[image]

closed 2023-01-31 by holla2040