bambulab/BambuStudio #1226: Ams wrong filament types

chuckkay, 2023-01-31:
i have 4 AMS Units 16 filaments, all labeled correctly on printer control screen.

when i sync on bambu studio, everything shows up correctly, i save 3mf and even save profile.  When i reopen Bambu studio, the filament colors are correct but filament types are wrong.

[image]
[image]

Latest versions of Windows, Bambu studio and firmware

bambu123, 2023-02-03:
Surprise! It behaves like this every time.
Does the old version have the same issue in you memory?

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei

TheCableGuy99, 2023-03-01:
Doesn't work across multiple printers.

bambu123, 2023-03-02:
Some filaments are not compatible with something printers.
Please provide details for the new issue.

TheCableGuy99, 2023-03-02:
I believe it's the same issue as I raised here and is supposed to be sorted. I've updated my own thread... if it is the same issue maybe these should be collated into a single issue? https://github.com/bambulab/BambuStudio/issues/1122#issuecomment-1451095316