bambulab/BambuStudio #1227: We Need from cura the "slicing tolerance> exclusive" Option. Its neccesary for printing screws and threads

Bangwa88, 2023-01-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

SaltWei, 2023-04-18:
@QingZhangBambu Please evaluate this feature request and see whether it is necessary

QingZhangBambu, 2023-04-21:
I think the function is similiar to xy-hole-compensation & xy-contour-compensation.
[image]

closed 2023-08-01 by QingZhangBambu