bambulab/BambuStudio #1229: Add an option to remove internal solid infill.

Babsobar, 2023-01-31:
**Is your feature request related to a problem? Please describe.**
I print things that have very small and slim walls, with Bambustudio, it will often add solid internal infill in parts that are unneeded.

**Describe the solution you'd like**
A checkbox to disable solid internal infill before slicing so that it is not integrated in the model

**Describe alternatives you've considered**
Using 0% infill and using modeled infill inside the model itself. Far from practical.

I have tried to make myself as clear as possible but english is not my mother tongue so if you have any questions or remarks about this issue don't hesitate

SaltWei, 2023-02-08:
[image]
Have you tried these two options？

Babsobar, 2023-02-08:
Yes, here is a screenshot presenting unwanted solid infill with two different setups:
[image]
[image]

SaltWei, 2023-02-22:
Could you share this 3MF to us for checking and debugging?

Babsobar, 2023-02-22:
Github does not allow the share of .3MF files, so here is a download link

[https://we.tl/t-10mzNhSDSY](https://we.tl/t-10mzNhSDSY?utm_campaign=TRN_TDL_05&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_05)

SaltWei, 2023-03-02:
[image]
Have you tried set this to be 0?

Babsobar, 2023-03-09:
Yes, that removes the top of the print, unwanted behavior.

DanielSchenk1337, 2023-07-07:
same here, there should be an option to deactivate the generation of inill in these areas, e.g. "disable solid internal infill".

How it looks in Bambu Studio:
[image]

How it should look like: (sliced in Cura)
[image]

Model: https://www.thingiverse.com/thing:1743145  "HIVE_Module_V5.stl"

aviatorstengel, 2023-12-02:
I am having the same issue when trying to print a building with small outer features. It is adding around an hour and a half to what should be a 4 hour print via Cura. See below.

[image]

[image]

Millagax, 2023-12-21:
> EDIT,SOLVED
> 
> **if you have the same problem as in the picture below, set _Minimum sparse infill threshold_ to 0 as SaltWei pointed out**
> 
> How it looks in Bambu Studio(solid infill not wanted):

I'm having the same issue, I want to get rid of solid infill but it's impossible. Someone marked it as solved, but that solution does not work. 

Can someone find a way to deactivate that annoying solid infill?

SaltWei, 2023-12-21:
> Can someone find a way to deactivate that annoying solid infill?

Could you share a 3mf so we can have a check? Maybe you met different issue with others

Millagax, 2023-12-21:
3MF file has nothing to do with this issue, besides, another user uploaded it and no aswer was given here. Is is a way to deactivate it or not?