bambulab/BambuStudio #1230: Add a 'Square' tool type for painting straight lines

CameronBigger, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
The problem my proposal would solve  is that there is currently no easy way to make a straight line of color across an arbitrary portion of a model. The closest you can come is to use the 'circle' tool to paint an arbitrary area - but it paints with a circle which does not provide a nice straight line.

**Describe the solution you'd like**
Add a 'Square' tool type. This would function almost exactly as the current 'Circle' option but would allow a square 'brush head' to enable coloring arbitrary lines on the model with great precision.

**Describe alternatives you've considered**
If you reduce the brush size to very small and use multiple passes of the circle tool, you can also make a straight line - but it is very tedious and takes a long time. Alternatively you can use the 'Height range' tool if your colored section cuts all the way across the model. But often time, I need to outline a rectangular shape that I can then fill in with color and so the Height Range tool isn't appropriate for that.

**Additional context**
Mockup screenshot is included.
[image]

retoid, 2024-04-03:
The ability to lock the cursor to an axis would be great as well. For example if you need a perfectly straight line going down a specific area of a model, lock cursor to Z axis and paint your line. Alternatively having a Line Tool where you click to determine point A, then click again to determine point B, where the Line would be drawn between each points.

DoTsTeR77, 2024-04-26:
this would be great.. we so need some additional painting tools.. I need a rectangle right now.. also a fille that could leave a border color on the fill area would be HUGE... like a STROKE on a selection in photohop..