bambulab/BambuStudio #1232: [Issue] Switching Print Settings "Others" Section Sub-Categories Not Displayed in Confirmation Window

BambuLabDiscord, 2023-02-01:
Changing some items under the Others settings does not specify the exact items in the print settings transfer or discard window. This leaves the user guessing at what item they are transferring to the newly selected profile or discarding.

Reproduction Steps:
1. Select any default print setting profile.
2. Under the Others tab, disable Prime Tower
3. Switch to a different default print setting
4. Observe only the "Others" setting and minimal data asking to be transferred or discarded.

clairmont32, 2023-02-01:
[image]

clairmont32, 2023-02-13:
Sorry for the late response. I'm on the latest Win 10 and Bambu Studio version and have this issue every time I switch print settings.

clairmont32, 2023-02-13:
Here's another screenshot from v1.4.2.13
[image]

SaltWei, 2023-02-15:
I see. "Enable" what....
Accept. Thank you very much for suggestions!
[image]