bambulab/BambuStudio #1233: Some color painted areas not correct color after slicing

CameronBigger, 2023-02-01:
**Describe the bug**
In a model rocket model, all of the fins tips were 'painted' black, mainly using the fill tool. As you can see in the 'prepare' shots, all four wing tips are showing black. However, after slicing, the preview shows two of the winters as being black and two as being white (they should all be black).

**3mf File for This Bug**
Attached.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the attached 3MF file featuring manually painted colors
2. Notice the color of all four wingtips should be black
3. Slice the model (I used 15% infill, grid and 0.20 layer height)

**Expected behavior**
All four wingtips would be black

**Actual results**
Two wingtips were black (correct) and two are partially white (wrong). Re-painted them several times but they always remain partially white after slicing.

**Screenshots**
Attached

**Desktop (please complete the following information):**
Mac OS Ventura (13.1)
Bambu Studio 1.4.2.13

[MercuryRedstone-bug report.3mf.zip](https://github.com/bambulab/BambuStudio/files/10561888/MercuryRedstone-bug.report.3mf.zip)
[image]

SaltWei, 2023-02-03:
[image]
That's seam preview

Lesani, 2023-02-03:
@CameronBigger Seams are where the printer starts and stops its loop with the black filament, this will not be white when printed

CameronBigger, 2023-02-03:
Thanks @Lesani. Gotcha. I almost wish the seam would be rendered in a very different color (maybe blue or orange or something in my model) than the filaments in use to make it more obvious this is the seam. 

I know about seams and usually put them on the "back" of objects like this rocket. But in this case, I didn't notice it. You are right: as soon as turned off seam visualization, I could see the wing tips were solid black.

Closing this as user misunderstanding :-)

closed 2023-02-03 by CameronBigger