bambulab/BambuStudio #1239: [Issue] Supports leave supported layer unfinished.

BambuLabDiscord, 2023-02-02:
First layer on top of supports is loose and not closed up let alone to say ugly.  It is like it has no skin at all but lines are spaced like a raft filling not a smooth closely printed surface.  I'm not expecting build plate quality but certainly better than an open window into the inners of the model

SaltWei, 2023-02-03:
Please share more information including 3MF or pictures

microwavesafe, 2023-07-04:
I seem to be having this issue as well, it has always been this way as the machine is new (less than 1 week old). I have tried 1.5 and 1.6 of Bambu studio and several different options for supports. The current project has 2 support solid layers and 1 interface layer height, and it still doesn't look like it is filling in the layer properly.

I've attached the 3mf file, I could remove the need for supports by orienting vertically, but this is a structural part and I have to orient the layers that way.
 
Good top surface
[image]

First layer on top of support
[image]

Project file zipped
[workbench_hanger.zip](https://github.com/bambulab/BambuStudio/files/11947965/workbench_hanger.zip)