bambulab/BambuStudio #1240: Bridge angle wrong

Lesani, 2023-02-03:
**Describe the bug**
Bridge angle is the same as other solid layers, instead of 90° to the bridge
Corners are not bridged correctly because they have no 2nd side to connect to yet, (or only a single strand of perimeter)

[image]

Prusaslicer for this reason (and maybe others) switches to 90° instead of 45° for the first bridge layer

[image]

**3mf File for This Bug**
[file.zip](https://github.com/bambulab/BambuStudio/files/10577157/file.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Slice

**Expected behavior**
complete bridge layer

**Desktop (please complete the following information):**
[image]

SaltWei, 2023-02-14:
Will be included in next release

SaltWei, 2023-03-02:
Included in 1.5.0 Public beta

closed 2023-03-02 by SaltWei