bambulab/BambuStudio #1241: HEX Color in Colorpicker

djalexz, 2023-02-03:
HEX Color in Colorpicker

It would be great if you could also enter the HEX value of the color when creating a custom filament.
Currently I help myself by manually editing the JSON file of the profile and entering the HEX value of the color there.

[image]

CameronBigger, 2023-02-03:
FYI: Maybe this is lacking in the Windows version, but there is an equivalent to this already available on the Mac version of Bambu Studio through the RGB Picker option.

[image]

djalexz, 2023-02-05:
Hi, Thank you for your answer.
I have this Problem with the Windows Version of Bambu Studio.
It would be great if the Windows Version has the Same color picker.

CameronBigger, 2023-04-01:
+1. @iandouglas if nobody already has, perhaps you should post this "custom color picker on printer touchscreen" as a separate suggestion?

walterwongbbl, 2023-04-03:
Hi.
We will provide some preset color choices.
and no longer allow users to edit custom colors.

Merlwynd, 2023-04-03:
No, do not take away the ability for users to edit custom colors! The original request was simply to allow users to enter a HEX value in the Windows version of Bambu.....

walterwongbbl, 2023-04-03:
> 不，不要剥夺用户编辑自定义颜色的能力！最初的请求只是允许用户在Windows版本的Bambu中输入十六进制值。

Sorry. my description is incomplete
we just don't support users editing custom colors for filaments with in AMS

walterwongbbl, 2023-08-02:
Please update the latest studio version.
we support user-defined color selectors

closed 2023-08-02 by walterwongbbl

nihilisticzebra, 2024-02-09:
months later still no fix common bambu