bambulab/BambuStudio #1242: Print head stays on top of model when print finishes (P1P

Hologos, 2023-02-03:
**Describe the bug**
The print head stays on top of the model for few seconds when the print finishes.

**3mf File for This Bug**
Nope

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

SaltWei, 2023-02-06:
Printer team will update end-gcode to fix this issue. Thank you very much!

eddydas, 2023-05-05:
Eagerly looking forward to have this fixed!

sodachen22, 2023-05-06:
Thanks for the feedback. Yes, it is a bug, and will be fixed.

Hologos, 2023-08-05:
@sodachen22 This seems to be fixed with latest version v1.7.2.51.

Hologos, 2024-02-16:
@SaltWei Hello, this can be closed as fixed.

closed 2024-02-18 by SaltWei