bambulab/BambuStudio #1243: K-Factor Setting for AMS Resets to 894685568,000

Laserstrahl, 2023-02-04:
Firmware P1P: 01.01.01.00
AMS: 00.00.05.75 (only using original Bambu Lab Basic PLA, which is recognized by the AMS)
Bambulab: 1.4.2.13

I'm using a P1P along with AMS.
When I restart the device and Bambulab the K-Factor for some Spools in the AMS are set to "894685568,000" without doing anything. This leads to a wrong flow rate and the print is messed up on the first layer.

Workaround:
Set the K-Factor manual under "Device" -> "Status" to "0" for every AMS Spool.

I can reproduce this bug by restarting the printer and restarting Bambulab.

StoneLiBambulab, 2023-02-13:
Hi @Laserstrahl ，
     Could you help to upload screen shot and log of Bambu Studio.
please refs to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log.

Thanks!

Stone Li

SaltWei, 2023-02-23:
Have comfirmed to be firmware bug, and will be fixed by firmware update.