bambulab/BambuStudio #1244: Cancel a failed print object on a plate when printing object with multiple part or multiple objects.

The-algar, 2023-02-04:
**Cancel a failed print object on a plate when printing object with multiple part or multiple objects.**
I miss the ability to cancel ONE specific object on a plate when printing object with multiple part or multiple objects.
Sometimes a part will detach from the plate and I want to be able to tell the printer to ignore that part but print everything else, so that I don't have to throw away what has been printed so far because a failed part caused an issue. 
Currently, the printer will just continue to extrude when there's nothing underneath it (because the object was detached from the plate during a print) and that will cause spaghetti that I have to manually remove if I want to finsih the print of the other parts/object.

jasoncouture, 2023-02-07:
This would be awesome when the toolhead knocks over 1 of 30 objects. We could do an online delete of the print, and avoid the spaghetti, as you stated!

Upvote: :+1:

j0nas, 2023-05-02:
This feature would be _immensely_ appreciated. OctoPrint has a [Cancel Object](https://plugins.octoprint.org/plugins/cancelobject/) plugin that is a must-have when printing multiple objects, perhaps one could draw inspiration from it?

As I am typing this: 
[image]

Fredyy90, 2023-07-31:
any updates on this feature? could really use it rightnow, to cancel 1 of 4 object on the bed.

[image]

SaltWei, 2023-07-31:
The firmware is under beta test. And will be published soon

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei